AlbrechtL/welle.io #721: [Raspberry pi 3B+] Output device Null

Ale268, 2022-03-07:
Hello there

Thak you for the great software, used it under windows and wanted it to now also run on a rpi3b+.

After finaly being able to build it using cmake it looked promising but it always fails using the output device "null".

It is running beside Moode Audio. Alsa is the used audio driver there. 
[image]

Attached a picture of the console out.

Let me know what other logs or informations you need!

Thank you beforehand and greetings from switzerland.

Seppklaus, 2022-03-16:
I have/had the same issue. Excellent reception with my Linux-Laptop and MacBook. No reception with the RPI3b+. I used the same rtl-sdr and the same antenna setup.
I then tried it with a RPI4 and same SD-Card and it works. I use the USB3 port on the RPI4.

I believe the problem is the power and quality of the RPI3 USB ports. Maybe it works with an additional power supply for the rtl-sdr but I don't have the cable to try it with my RPI3.

Maybe some one else can confirm that the issues with the RPI3 are due to the power of the USB port - or not.

AlbrechtL, 2022-08-21:
Thanks for reporting. The issue is that welle.io doesn't find an audio output interface. Did you use the latest Raspberry OS?

Ale268, 2022-08-22:
Hello there

Thank you for Answering. 
I am trying to get it running on a RPI3B+ which is running moodeaudio as its base operating System. 
Moode is just an adapted Raspbian though. 

[image]
Here are all the related versions.

The audio output device is an I2S Board (Hifiberry DAC+ DSP).

Ale268, 2022-08-22:
> 

> However, I installed it simply with:
> sudo apt install welle.io

When I try to do so it won't find anything
[image]

AlbrechtL, 2022-12-16:
In the latest Raspberry OS welle.io is included.

AlbrechtL, 2022-12-16:
For now I'm closing this issue.

closed 2022-12-16 by AlbrechtL

Ale268, 2023-06-28:
I have now installed everything again but the output device is still null.