AlbrechtL/welle.io #727: Use another form of input for rtl_tcp settings

SergeantSerk, 2022-03-30:
Using the current interface to select the IP address numbers and port is incredibly difficult. The rtl_tcp port that is currently running for me is 9999 and I'm having to scroll all the way from 1234 which is very time consuming and not user friendly.

Perhaps a simple text box to input and then do validation would be better?

AlbrechtL, 2022-08-21:
You are right. The spinner for port setting is not really user friendly.

As a workaround you can lower you port number.

stonedDiscord, 2022-09-01:
it works great on touch displays but with a mouse they are a real pain

ideally this should only be a spinner on android and a classic text box for all the other platforms

andimik, 2022-09-02:
And what about splitting the 4 digits into 4 spinners?

SergeantSerk, 2022-09-02:
Make that 5 digits since max port is 65535 :)

AlbrechtL, 2024-10-09:
In https://github.com/AlbrechtL/welle.io/commit/466be3df9894ca9f2b4a075902f2b16f42d4ca62 I changed to rtltcp settings (see screenshot). Now you can easily use big port numbers.

[image]

Feel free to test the latest unstable build: https://welle-io-nightlies.albrechtloh.de/20241009_466be3df_Windows_welle-io-setup_x64.exe

closed 2024-10-09 by AlbrechtL