AlbrechtL/welle.io #730: Quits unexpectedly every time the app is closed - MacOS

ambanmba, 2022-04-06:
Each time the app is closed the following problem report is produced:

`-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               welle-io [30717]
Path:                  /Applications/welle.io.app/Contents/MacOS/welle-io
Identifier:            com.yourcompany.welle-io
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2022-04-06 18:01:42.9460 +1000
OS Version:            macOS 12.3.1 (21E258)
Report Version:        12
Bridge OS Version:     6.4 (19P4243)
Anonymous UUID:        9C9843D4-24BD-909C-1228-26E2E2FF698D

Time Awake Since Boot: 440000 seconds

System Integrity Protection: enabled

Crashed Thread:        6

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

ID   Vend/Dev
ae8e 67df1002
Seconds Ago   ID   Type
   440000.0   ae8e Attach

Application Specific Information:
abort() called
terminating

Thread 0::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	    0x7ff81ecb90ea __ulock_wait + 10
1   libsystem_pthread.dylib       	    0x7ff81ecf5c95 _pthread_join + 358
2   libc++.1.dylib                	    0x7ff81ec5c986 std::__1::thread::join() + 24
3   welle-io                      	       0x10aac4712 CRTL_TCP_Client::stop() + 66
4   welle-io                      	       0x10aac36ba CRTL_TCP_Client::~CRTL_TCP_Client() + 138
5   welle-io                      	       0x10aac382e CRTL_TCP_Client::~CRTL_TCP_Client() + 14
6   welle-io                      	       0x10aadf3e2 CRadioController::closeDevice() + 194
7   welle-io                      	       0x10aaded88 CRadioController::~CRadioController() + 56
8   welle-io                      	       0x10aaceccc main + 5644
9   dyld                          	       0x11a15751e start + 462

Thread 1:: CAudioThread
0   libsystem_kernel.dylib        	    0x7ff81ecbe0aa poll + 10
1   QtCore                        	       0x10da7f8ee qt_safe_poll(pollfd*, unsigned int, timespec const*) + 222
2   QtCore                        	       0x10da810bc QEventDispatcherUNIX::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 812
3   QtCore                        	       0x10da1e0c7 QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 471
4   QtCore                        	       0x10d85dc2c QThread::exec() + 140
5   QtCore                        	       0x10d85eb3a 0x10d83c000 + 142138
6   libsystem_pthread.dylib       	    0x7ff81ecf44e1 _pthread_start + 125
7   libsystem_pthread.dylib       	    0x7ff81eceff6b thread_start + 15

Thread 2:: AMCP Logging Spool
0   libsystem_kernel.dylib        	    0x7ff81ecb79b6 semaphore_wait_trap + 10
1   caulk                         	    0x7ff8277ca2e0 caulk::mach::semaphore::wait_or_error() + 16
2   caulk                         	    0x7ff8277b2102 caulk::concurrent::details::worker_thread::run() + 36
3   caulk                         	    0x7ff8277b1dc6 void* caulk::thread_proxy<std::__1::tuple<caulk::thread::attributes, void (caulk::concurrent::details::worker_thread::*)(), std::__1::tuple<caulk::concurrent::details::worker_thread*> > >(void*) + 41
4   libsystem_pthread.dylib       	    0x7ff81ecf44e1 _pthread_start + 125
5   libsystem_pthread.dylib       	    0x7ff81eceff6b thread_start + 15

Thread 3:: QDBusConnectionManager
[... 363 lines cut ...]
      "deploymentId" : 240000008
    }
  ],
  "experiments" : [

  ]
}
}

Model: Macmini8,1, BootROM 1731.100.130.0.0 (iBridge: 19.16.14243.0.0,0), 6 processors, 6-Core Intel Core i7, 3.2 GHz, 64 GB, SMC 
Graphics: Intel UHD Graphics 630, Intel UHD Graphics 630, Built-In
Display: VZ27A, 2560 x 1440 (QHD/WQHD - Wide Quad High Definition), MirrorOn, Online
Graphics: AMD Radeon RX 580, AMD Radeon RX 580, PCIe, 8 GB
Display: VZ27A, 2560 x 1440 (QHD/WQHD - Wide Quad High Definition), MirrorOn, Online
Display: MZ27AQ, 2560 x 1440 (QHD/WQHD - Wide Quad High Definition), Main, MirrorOff, Online
Memory Module: BANK 0/ChannelA-DIMM0, 32 GB, DDR4, 2667 MHz, 859B, CT32G4SFD8266.C16FB
Memory Module: BANK 2/ChannelB-DIMM0, 32 GB, DDR4, 2667 MHz, 859B, CT32G4SFD8266.C16FB
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x7BF), wl0: Jul 12 2021 19:26:30 version 9.30.464.0.32.5.76 FWID 01-45ccefcd
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Ethernet, Ethernet, en0
Network Service: Wi-Fi, AirPort, en1
PCI Card: pcic0a9,540a, NVM Express Controller, Thunderbolt@197,0,0
PCI Card: AMD Radeon RX 580, Display Controller, Thunderbolt@71,0,0
PCI Card: pci1002,aaf0, Audio Device, Thunderbolt@71,0,1
`

AlbrechtL, 2022-08-21:
Thanks for reporting!

AlbrechtL, 2024-10-08:
I'm closing this issue because I'm not sure if it is still relevant. If it still occurs in the latest version, we can reopen it.

closed 2024-10-08 by AlbrechtL