AlbrechtL/welle.io #732: Favourite stations

sierragls, 2022-05-18:
It it possible to save favourite stations? This will save having to scroll all the way to find the station I want to listen to on android car stereo.

andimik, 2022-05-19:
Yes, like in a browser just press the asterisk icon.

sierragls, 2022-05-19:
Can this be done on android? I haven’t seen an asterisk. Tried holding my finger on the station but nothing.

andimik, 2022-05-19:
You just need to click on the icon. Then it will become yellow instead of white. 

But I confirm that there is a bug.

The favorite list stays empty on Android. I have never tested it before.

[image]
[image]

sierragls, 2022-05-19:
Mine is different to that. There is no star. 

[image]
 
[image]

sierragls, 2022-05-19:
Got it now I had an older version  downloaded it, you are right favourites don’t work. 

There is another issue when i minimise the app to do something else the signal goes so the app has to be constantly on screen for it to work, are there any settings to change this?

closed 2022-05-19 by sierragls