AlbrechtL/welle.io #734: [Android] RAW file not supported

andimik, 2022-05-19:
Using a raw file under Android is not supported. After selecting a raw file, the entry "Selected file:" just says the German text 'Derzeit unter Android angezeigt' which is 'Currently shown under Android' in English, that comes from this line:

https://github.com/AlbrechtL/welle.io/blob/0aacf578d77d79212483cd6fce436fb66523f63f/src/welle-gui/QML/settingpages/RawFileSettings.qml#L71
[image]
[image]

AlbrechtL, 2022-12-16:
Thanks for reporting! I think the sentence was a quick hack to get it running under Android.

AlbrechtL, 2024-10-08:
I'm closing this issue because support RAW files under Android is out of scope. If somebody sends a pull request I will review it.

closed 2024-10-08 by AlbrechtL