AlbrechtL/welle.io #741: Can not find list of endpoints for welle-cli

SzymonZy, 2022-06-12:
Hello

I have found on https://forums.openpli.org/topic/56134-dab-radio-reception/page-19 informations that it is possible to tune and stream via cli webserver - how ever this format is always failing with:
Malformed request:

Is there any documentation for that? I have tried: 
192.168.1.34:1234/mp3/11A/0x3211
for channels shown in gui:
[image]

My goal is to build m3u and play remotely in Kodi

mpbraendli, 2022-06-16:
I believe the feature you're looking for wasn't yet merged. Done just now, see here for details:
https://github.com/AlbrechtL/welle.io/pull/740

closed 2022-06-16 by mpbraendli

comcloudway, 2022-06-16:
As I just stated in #740 my implementation is not the same as the one proposed over on the forum.
I exposed an endpoint at `/mp3-channel/<channel>/<program>`.
Keep in mind that my PR has been merged into the [next](https://github.com/AlbrechtL/welle.io/tree/next) branch, so you will probably have to wait for the next version to be able to use the feature

SzymonZy, 2022-06-16:
Ok. Thank You cery much. Just one question. Is three some powersaving mode.
When there is no playback does tuner unlock from frequency and power down?
I ask because my gets very hot even not used.

comcloudway, 2022-08-19:
I'm not sure, but I doubt there is, given that this would require you to retune every time you return from power saving mode. 
You might want to file a separate issue for this.