AlbrechtL/welle.io #744: [Android] Service duplicates in Channellist

dhwz, 2022-06-23:
In the new version service duplicates received via multiple ensembles are no longer visible nor can be selected.
e.g. "Bayern 1 Main" received and scanned on 8C but stronger signal is on channel 11D. But only the 8C service is listed, in the old Android version both were listed an the stronger signal could be selected.

Can we get that functionality back (at least optionally in the settings)?

AlbrechtL, 2022-12-16:
This is a corner case for people in Bavaria. I removed it because I had to many complains from Bavarian peoples. What you can do is to clear the channel list, switch in the expert mode and then tune to each channel separately. Start with 11D first. Then you should have it correctly.

closed 2022-12-16 by AlbrechtL

dhwz, 2022-12-17:
Sehr schade denn das ist leider nur ein Workaround und keine Lösung.
Es betrifft auch nicht nur Leute in Bayern, der Bundesmux 2 sendet auf unterschiedlichen Kanälen, bei Doppelempfang wird ggf. zuerst der schwächere Mux gescannt.

Könnte man nicht alle Sender abscannen lassen und die Empfangsstärke merken und dann am Ende des Suchlaufs nur die schwächeren Duplikate verwerfen?

So machen das ja auch viele Hardwaregeräte.

AlbrechtL, 2023-02-18:
If you like you can rise up a pull request if you want to implement this logic :-)