AlbrechtL/welle.io #747: An easy way to make m3u files for all your radio stations you can receive.

rocus, 2022-07-31:
I have some audio streamers on my network (mostly UPNP streamers and MPD on raspberries) that can play audio streams from streaming hosts like welle-cli (started with -w option). Welle-cli can generate a mux.json file for the channel you have chosen. For all channels you could use these mux.json files and generate a m3u file for every radio station you can receive.
Here I describe an easier way to generate these files.

Start by using welle.io (installed together with welle-cli) to scan all channel frequencies.
This generates in your config directory (in Debian ~/.config/welle.io) a file  welle.io.conf.
Convert this file to a json file with all radio stations (with name, sid and channel):

grep stationListSerialize welle.io.conf | sed 's/stationListSerialize=\"// ; s/]\"/]/ ; s/\\//g'  > stations.json

To create all m3u files I made a small python script:

#!/usr/bin/python3
import json

def gen_m3u ( station ):
    name = station [ 'stationName' ] . rstrip () 
    sid  = station [ 'stationSId'  ]
    chn  = station [ 'channelName' ]
    print  ( name , hex(sid) , chn )
    with open ( 'DAB '+name+'.m3u' , 'w' ) as p:
        p.write ( '#EXTM3U\n#EXTINF:-1,1 '+name+'\nhttp://10.0.2.217:8888/mp3/'+hex(sid)+'\n' )
        p.close 

with open('stations.json') as f:
    stations = json.load(f)
    for station in stations:
        gen_m3u ( station )

If there are several channels you can receive, you can only play those m3u playlist files that are in the mux of your current channel: you can change the channel in the webinterface with which you started welle-cli .
To let the channel change automatically is not possible with the current welle-cli.
I describe a hack of mine to make this possible as a comment on issue 712.

mpbraendli, 2022-08-18:
Nice! For improved readability you can use [f-strings](https://fstring.help/cheat/)

```python
 with open(f'DAB{name}.m3u', 'w') as p:
        p.write(f'#EXTM3U\n#EXTINF:-1,1 {name}\nhttp://10.0.2.217:8888/mp3/{hex(sid)}\n')
```

Is there a better place to keep this hint than in an issue? Maybe in the README or wiki?

macmpi, 2022-09-10:
Very interesting hint indeed.
@AlbrechtL have you considered packaging [welle-cli](https://github.com/AlbrechtL/welle.io#welle-cli) within a [MPD input source plugin](https://mpd.readthedocs.io/en/latest/plugins.html#input-plugins) for seamless DAB+ playback on this very versatile & popular environment: user could leverage many exiting [GUI clients](https://www.musicpd.org/clients/) on any platform to controll it?

AlbrechtL, 2022-12-16:
@rocus Thanks. You are welcome to create a pull request with you small script.

@macmpi I lot of thinks are possible :-). If somebody want to code such a plugin - go ahead!

closed 2022-12-16 by AlbrechtL

Lamarqe, 2024-08-31:
@rocus , @macmpi : https://github.com/Lamarqe/mpdcast-dab should solve your use-case. Its based on welle.io.