AlbrechtL/welle.io #753: Sign your Android apk

tenzap, 2022-08-24:
For the latest release, I believe you simply published the APK built by the 'nightly' job.

Please note that this APK is not signed. Actually it is, but with the developer key, and it is valid only for 1 year iirc, hence it can be installed only during that period.

Would you please sign the APK with your keys? Either by updating the Travis job to have it done on each commit, or by signing at least the official releases when you punish them?

Instructions can be found here for qt6.
https://doc.qt.io/qt-6/deployment-android.html#building-the-android-application

Extract from that page:

> Under the hood, your Qt code is built and copied to the $ANDROID_BUILD_DIR, then Gradle is used to build the Java code and package the application. If an APK intended for release is built, then it should be signed with jarsigner and aligned with zipalign.

Btw, for the older qt5, see https://github.com/AlbrechtL/welle.io/pull/679

AlbrechtL, 2022-12-16:
Is there an option to sign the nightly build via Github actions? I mean there is the function secrets. Feel free to take a look into it! For sure I can place a nightly build private key inside the Github action secrets.

AlbrechtL, 2024-10-08:
Closed because we are discussing it here https://github.com/AlbrechtL/welle.io/issues/814.