AlbrechtL/welle.io #761: Software Selectable Bias Tee on RTL-SDR Blog V3

maxzz69, 2022-09-05:
Hi,

I'm using Windows11 with welle.io 2.4. Is it possible to use the Software Selectable Bias Tee on RTL-SDR Blog V3.
It's described on https://www.rtl-sdr.com/rtl-sdr-blog-v-3-dongles-user-guide/  and I could download a zipfile and run a batchfile everytime before startin welle.io, but it would be nice if this would be possible within welle.io.

kind regards.
Richard

gvanem, 2022-09-06:
On which OS?
Try running `rtl_biast -d 0 -b 1` (or similar) before starting welle-io. But it depends on which `rtlsdr.dll` you use.

maxzz69, 2022-09-06:
Hi,

I use Windows 11 and "rtl_biast -b 1 -d 1" works for me, but it would be nice if this could be build-in.

Kind regards,
Richard

AlbrechtL, 2022-12-16:
Thanks for the idea. Pull request are welcome!

AlbrechtL, 2024-10-09:
I'm not sure if we can include the RTL-SDR blog V3 dongle software bias tee into welle.io. welle.io already supports bias tee but because of this request it seems that this not the correct once for the software bias tee.

Inside the original RTLSDR driver I can't find a software bias tee. I was trying to use the RTL-SDR blog driver, but the driver was running unstable in welle.io (see https://github.com/rtlsdrblog/rtl-sdr-blog/issues/56).

Currently, the only option seems to be to follow the explanation on https://www.rtl-sdr.com/rtl-sdr-blog-v-3-dongles-user-guide/
_**Forcing the Bias Tee to be Always On**
If you are using our [RTL-SDR-Blog driver branch](https://github.com/rtlsdrblog/rtl-sdr-blog) you can force the bias tee to be always on by setting a flag in the EEPROM. The rtl_eeprom command is "rtl_eeprom -b 1". Run the opposite command "rtl_eeprom -b 0" to disable the forced bias tee._

closed 2024-10-09 by AlbrechtL