AlbrechtL/welle.io #765: Mac : output device

joeriblauw, 2022-09-17:
I love this piece of SW to listen to DAB+ on my Mac, and only wish I could choose a different output device.
Welle plays to the built-in speakers and nothing else. Choosing a different output device does not have an effect.
I guess (not a programmer) that the output was low-level hard-coded to the Mac default device = always the speakers.

Could this be the result of using MacPorts to run Welle.io on my Mac ?
Does anyone have a suggestion to overcome this issue?

thanks! J.

aerogus, 2022-09-22:
You could use a 3rd party software like Loopback https://rogueamoeba.com/loopback/ to route every apps to the output you want. It's not free, it's generic, it doesn't "overcome this issue" but it could help :)

ambanmba, 2022-12-05:
On macOS you can't change the output device after welle.io is started. Select the output device first, then start welle.io.

AlbrechtL, 2022-12-16:
We already have the discussion here https://github.com/AlbrechtL/welle.io/issues/116. I still think it is worth to try it. Especially with the new QT6 multimedia classes.

closed 2022-12-16 by AlbrechtL