AlbrechtL/welle.io #771: welle.io_2.4_Linux_x86_64.AppImage not working in opensuse Leap 15.4

seuchato, 2022-11-22:
Hi
I have a daily updated install of opensuse 15.4. I do have this sdr stick:
` Bus 001 Device 023: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T`
It is the blue "nooelec NESDR Mini 2+ "820T2 SDR & DVB-T" stick. Using gqrx, it is recognised  plots a waterfall spectrum and plays sound. I'd say it works. But I'd like DAB+ => enter welle.io:
 
I downloaded welle.io_2.4_Linux_x86_64.AppImage and started it.
No burger menu, no scan option and off course no stations, see attached screenshot:
[image]
How can this be fixed?
greez
chris

andimik, 2022-11-23:
Assuming you are from a German speaking country (due to the screenshot), please use the expert settings and scan manually 5C (if you are from Germany) or 12C (Switzerland).

If you are from Austria, please note that the Bundesmux is not available in some regions. Therefore, check https://dabplus.at/

Anyway, please add the spectrum diagram and try manual gain settings or change the (very very cheap) RTL Stick.

seuchato, 2022-11-23:
Thanks, however I have not found a clue to start welle.io in expert mode. Maybe, you could point me to some reference?
Meanwhile I built a rpm and started welle.io from it. Here is a screenshot:
[image]
About the same as before, but now I got the prominent red message saying there is no device found, it uses standard device.

welle-cli  seems to confirm at least a problem with invoking the rtl device. Here is what I get from:
`welle-cli -c 12A
welle-cli unknown
InputFactory:Input device:auto
RTL_SDR: Open rtl-sdr
**_RTL_SDR: Found 1 devices_**. Uses the first working one
usb_claim_interface error -6
**_RTL_SDR:  Opening rtl-sdr failed_**
SoapySDR: No usable SDR device found
Error: No valid device found use Null device instead.
OFDM-processor:restart
Wait for sync
^C
`
Somehow the highlighted parts are contradicting?!

seuchato, 2022-11-23:
Here is additional info following up ""

======= as code tags do not seem to work =======
lsmod | grep sdr
rtl2832_sdr            40960  0
videobuf2_vmalloc      20480  1 rtl2832_sdr
videobuf2_v4l2         28672  1 rtl2832_sdr
videobuf2_common       69632  4 rtl2832_sdr,videobuf2_vmalloc,videobuf2_v4l2,videobuf2_memops
videodev              274432  3 rtl2832_sdr,videobuf2_v4l2,videobuf2_common
usbcore               323584  11 rtl2832_sdr,xhci_hcd,ehci_pci,dvb_usb_v2,usbhid,dvb_usb_rtl28xxu,usb_storage,ehci_hcd,xhci_pci,uas
===========================================

unplugged the stick and then:

======= as code tags do not seem to work =======
dmesg -T | grep -Ei rtl
[Wed Nov 23 12:08:02 2022] usb 1-11: Product: RTL2838UHIDIR
[Wed Nov 23 12:08:02 2022] usb 1-11: dvb_usb_v2: found a 'Realtek RTL2832U reference design' in warm state
[Wed Nov 23 12:08:02 2022] dvbdev: DVB: registering new adapter (Realtek RTL2832U reference design)
[Wed Nov 23 12:08:02 2022] rtl2832 8-0010: Realtek RTL2832 successfully attached
[Wed Nov 23 12:08:02 2022] usb 1-11: DVB: registering adapter 0 frontend 0 (Realtek RTL2832 (DVB-T))...
[Wed Nov 23 12:08:02 2022] dvbdev: dvb_create_media_entity: media entity 'Realtek RTL2832 (DVB-T)' registered.
[Wed Nov 23 12:08:02 2022] rtl2832_sdr rtl2832_sdr.2.auto: Registered as swradio0
[Wed Nov 23 12:08:02 2022] rtl2832_sdr rtl2832_sdr.2.auto: Realtek RTL2832 SDR attached
[Wed Nov 23 12:08:02 2022] rtl2832_sdr rtl2832_sdr.2.auto: SDR API is still slightly experimental and functionality changes may follow
[Wed Nov 23 12:08:02 2022] rc rc0: Realtek RTL2832U reference design as /devices/pci0000:00/0000:00:14.0/usb1/1-11/rc/rc0
[Wed Nov 23 12:08:02 2022] rc rc0: lirc_dev: driver dvb_usb_rtl28xxu registered at minor = 0, raw IR receiver, no transmitter
[Wed Nov 23 12:08:02 2022] input: Realtek RTL2832U reference design as /devices/pci0000:00/0000:00:14.0/usb1/1-11/rc/rc0/input43
[Wed Nov 23 12:08:02 2022] usb 1-11: dvb_usb_v2: 'Realtek RTL2832U reference design' successfully initialized and connected
===========================================

restarted welle.io, now the red message is gone, same screenshot as on Nov 22: no burger menu, no stations, no option selectable.
"welle-cli -c 12A" still prints "usb_claim_interface error -6 //  RTL_SDR:  Opening rtl-sdr failed"

So no progress

seuchato, 2022-11-23:
Some success:
copied "https://raw.githubusercontent.com/osmocom/rtl-sdr/master/rtl-sdr.rules" to /etc/udev/rules.d, unplugged the stick, did "udevadm control --reload-rules && udevadm trigger", plugged the stick back in, now I get this:

###########################################
welle-cli -c 12A
welle-cli unknown
InputFactory:Input device:auto
RTL_SDR: Open rtl-sdr
RTL_SDR: Found 1 devices. Uses the first working one
Detached kernel driver
Found Rafael Micro R820T tuner
RTL_SDR:  Opening rtl-sdr device0
[R82XX] PLL not locked!
RTL_SDR:  Supported gain values29
RTL_SDR:  gain 49.6
RTL_SDR:  gain 48
RTL_SDR:  gain 44.5
RTL_SDR:  gain 43.9
RTL_SDR:  gain 43.4
....
###########################################

So, it seems there was a problem in the previous udev rules. However, the gui "welle.io" is still as per previous posts: No burger meno, no options selectable (apart from "favourites/all stations", Play and volume, which I forgot to mention in previous posts).

andimik, 2022-11-23:
You will find the (invisible) right of the volume icon.

There please use the expert settings and follow my hints above.

Or you use rtl_tcp on port 1234.

seuchato, 2022-11-23:
I am listening to DAB right now. The key was the rules file plus finding the menu to switch into expert mode. 
There is a button(?) and a menu not visible on my box:
Expert settings: right of volume pictogram:
[image]
Outmost left of text "welle.io" 
[image]
This one toggles station list.

Why are these two without any pictograms? The one on the right side should show
[image]
ie. three vertically arranged dots, and the one on the left should show 
[image]
How do I get these pictograms where they should be? By the way, this is on both the app image and the rpm based install.

andimik, 2022-11-23:
See #725 and #713

seuchato, 2022-11-23:
I am listening to DAB right now. The key was the rules file plus finding the menu to switch into expert mode. So big fat thanks Andimik.
I am yet to apply the patch. I guess the current source tarball has that patch already.
Any interest in getting my rpm here (provided, it works on other machines too)?

seuchato, 2022-11-24:
Applied the patch, rebuilt the rpm installed with rpm -Uv, verified the new version is installed, but burger and three dot menu pictos still are not appearing.
I develop apps under GT-designer myself. Will try to get it imported and see from there where it fails.

seuchato, 2022-11-24:
Have it sorted out. Rpm installs and runs nicely. Thanks for your support.
I repeat my offer to share the rpm here, if wanted.

closed 2022-11-30 by seuchato