AlbrechtL/welle.io #775: Unable to launch GUI

Ale268, 2022-12-27:
Hello there 

I Installed it on a raspberry pi 3b+ with the official command.

Sadly when I try to start the GUI i receive the following error which I am unable to resolve:
```
pi@ReVoxcortex:~ $ export QT_DEBUG_PLUGINS=1
pi@ReVoxcortex:~ $ Display=:0 welle-io                                                                                                                                                                                               
2022-12-28T00:39:07.942 Debug: QFactoryLoader::QFactoryLoader() checking directory path "/usr/lib/aarch64-linux-gnu/qt5/plugins/platforms" ...                                                                                       
2022-12-28T00:39:07.945 Debug: QFactoryLoader::QFactoryLoader() looking at "/usr/lib/aarch64-linux-gnu/qt5/plugins/platforms/libqeglfs.so"                                                                                           
2022-12-28T00:39:07.947 Warning: Found metadata in lib /usr/lib/aarch64-linux-gnu/qt5/plugins/platforms/libqeglfs.so, metadata=                                                                                                      
{                                                                                                                                                                                                                                    
    "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",                                                                                                                                                         
    "MetaData": {                                                                                                                                                                                                                    
        "Keys": [                                                                                                                                                                                                                    
            "eglfs"                                                                                                                                                                                                                  
        ]                                                                                                                                                                                                                            
    },                                                                                                                                                                                                                               
    "archreq": 0,                                                                                                                                                                                                                    
    "className": "QEglFSIntegrationPlugin",                                                                                                                                                                                          
    "debug": false,                                                                                                                                                                                                                  
    "version": 331520                                                                                                                                                                                                                
}                                                                                                                                                                                                                                    
                                                                                                                                                                                                                                     
                                                                                                                                                                                                                                     
2022-12-28T00:39:07.950 Debug: Got keys from plugin meta data ("eglfs")                                                                                                                                                              
2022-12-28T00:39:07.950 Debug: QFactoryLoader::QFactoryLoader() looking at "/usr/lib/aarch64-linux-gnu/qt5/plugins/platforms/libqlinuxfb.so"                                                                                         
2022-12-28T00:39:07.952 Warning: Found metadata in lib /usr/lib/aarch64-linux-gnu/qt5/plugins/platforms/libqlinuxfb.so, metadata=                                                                                                    
{                                                                                                                                                                                                                                    
    "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",                                                                                                                                                         
    "MetaData": {                                                                                                                                                                                                                    
        "Keys": [                                                                                                                                                                                                                    
            "linuxfb"                                                                                                                                                                                                                
        ]                                                                                                                                                                                                                            
    },                                                                                                                                                                                                                               
    "archreq": 0,                                                                                                                                                                                                                    
    "className": "QLinuxFbIntegrationPlugin",                                                                                                                                                                                        
    "debug": false,                                                                                                                                                                                                                  
    "version": 331520                                                                                                                                                                                                                
}                                                                                                                                                                                                                                    
                                                                                                                                                                                                                                     
                                                                                                                                                                                                                                     
2022-12-28T00:39:07.954 Debug: Got keys from plugin meta data ("linuxfb")                                                                                                                                                            [... cut ...]
pi@ReVoxcortex:~ $ ldd
ldd: missing file arguments                                                                                                                                                                                                          
Try `ldd --help' for more information.                                                                                                                                                                                               
pi@ReVoxcortex:~ $

```

AlbrechtL, 2023-02-18:
Which OS do you are running at the RPi? Did you install it via `apt`?

Ale268, 2023-02-20:
I'm running on Raspbian.
The derivate is moodeaudio. 
Yes indeed I installed through apt.

tenzap, 2023-03-06:
`Display=:0` ? On linux PC I use `DISPLAY=:0` (ie upper case)

Ale268, 2023-06-28:
Sadly the SD in the pi died. After setting the whole thing up again and using the DISPLAY variable as you suggested it works now. Thank you very much.

Sadly i can't run any station search but maybe i'll find out why :)

AlbrechtL, 2024-10-08:
Closed because of inactivity.

closed 2024-10-08 by AlbrechtL