AlbrechtL/welle.io #776: welle-cli leaks memory

alexmyczko, 2023-02-01:
I have a user reporting memory leak, more details here:

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1030236

Thank you for looking into it...

dieterdeyke, 2023-02-03:
I was trying to use welle-cli as a sub-process to an interactive text ui. Leaving stdin unchanged welle-cli stole key strokes intended for the parent process. Connecting stdin to /dev/null made welle-cli block and stall, and connecting stdin to /dev/zero made welle-cli leak memory very fast. Do you have any tips how I can achieve my goal?

alexmyczko, 2023-02-03:
@dieterdeyke did you try to use it as documented in /usr/share/doc/welle.io/ ?
https://salsa.debian.org/debian-hamradio-team/welle.io/-/blob/master/debian/README.Debian

dieterdeyke, 2023-02-03:
Alex Myczko ***@***.***> writes:

> @dieterdeyke did you try to use it as documented in /usr/share/doc/welle.io/ ?
> https://salsa.debian.org/debian-hamradio-team/welle.io/-/blob/master/debian/README.Debian

No, that is not the use model I want to use. I would like to run
welle-cli as a subprocess to an interactive text ui.

I commented on https://github.com/AlbrechtL/welle.io/issues/776:

I was trying to use welle-cli as a sub-process to an interactive text
ui. Leaving stdin unchanged welle-cli stole key strokes intended for the
parent process. Connecting stdin to /dev/null made welle-cli block and
stall, and connecting stdin to /dev/zero made welle-cli leak memory very
fast. Do you have any tips how I can achieve my goal?

-- 
Dieter Deyke
***@***.***
Get my Gnupg key:
gpg --keyserver keys.gnupg.net --recv-keys B116EA20

dieterdeyke, 2023-02-03:
Just to close this issue, I have written a C wrapper using ptys to solve my problem:

#! /usr/bin/tcc -run

/* Author: Dieter Deyke <dieter.deyke@gmail.com> */
/* Time-stamp: <2023-02-03 14:48:36 deyke> */

#define _XOPEN_SOURCE 500

#include <fcntl.h>
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <unistd.h>

int main(int argc, char **argv)
{

    char *channel;
    char *program;
    char *slave_name;
    int master_fd;

    if (argc != 2 || strncmp(argv[1], "dab://", 6)) {
        fprintf(stderr, "Usage: dabplayer <url>\n");
        exit(1);
    }
    channel = argv[1] + 6;
    if (!(program = strchr(channel, '/'))) {
        fprintf(stderr, "Usage: dabplayer <url>\n");
        exit(1);
    }
    *program++ = 0;

    if ((master_fd = open("/dev/ptmx", O_RDWR, 0600)) < 0) {
        perror("open(\"/dev/ptmx\"");
        exit(1);
    }
    if (!(slave_name = ptsname(master_fd))) {
        perror("ptsname");
        exit(1);
    }
    if (unlockpt(master_fd)) {
        perror("unlockpt");
        exit(1);
    }

    close(0);
    if (open(slave_name, O_RDWR) != 0) {
        fprintf(stderr, "stdin fd is not 0\n");
        exit(1);
    }

    close(1);
    if (open("/dev/null", O_RDWR) != 1) {
        fprintf(stderr, "stdout fd is not 1\n");
        exit(1);
    }

    close(2);
    if (open("/dev/null", O_RDWR) != 2) {
        fprintf(stderr, "stderr fd is not 2\n");
        exit(1);
    }

    execl("/usr/bin/welle-cli", "/usr/bin/welle-cli", "-c", channel, "-p", program, 0);
    perror("execl");

    return 0;
}

alexmyczko, 2023-02-13:
@dieterdeyke I haven't tried your code, but if you think it's worth to be added to upstream, why not suggest so (is it maybe useful to others)? Shall I close the issue?

closed 2023-02-13 by alexmyczko