AlbrechtL/welle.io #777: Install problem with MAC ARM OS Ventura 13

Misterbianchi, 2023-02-06:
Hi Everyone, 

I followed the guide to install welle.io

`brew tap AlbrechtL/welle_io https://github.com/AlbrechtL/welle.io`

`brew install AlbrechtL/welle_io/welle.io`

But I get the following error 

> Failed executing: cmake . -DCMAKE_INSTALL_PREFIX=/opt/homebrew/Cellar/welle.io/2.4 -DCMAKE_INSTALL_LIBDIR=lib -DCMAKE_BUILD_TYPE=Release -DCMAKE_FIND_FRAMEWORK=LAST -DCMAKE_VERBOSE_MAKEFILE=ON -Wno-dev -DBUILD_TESTING=OFF -DCMAKE_OSX_SYSROOT=/Library/Developer/CommandLineTools/SDKs/MacOSX13.sdk -DRTLSDR=1 -DSOAPYSDR=0 -DWITH_APP_BUNDLE=0 -DBUILD_WELLE_CLI=1 (BuildError)
	from /opt/homebrew/Library/Homebrew/formula.rb:2462:in `open'
	from /opt/homebrew/Library/Homebrew/formula.rb:2462:in `system'
	from /opt/homebrew/Library/Taps/albrechtl/homebrew-welle_io/HomebrewFormula/welle.io.rb:23:in `install'
	from /opt/homebrew/Library/Homebrew/build.rb:177:in `block (3 levels) in install'
	from /opt/homebrew/Library/Homebrew/utils.rb:605:in `with_env'
	from /opt/homebrew/Library/Homebrew/build.rb:139:in `block (2 levels) in install'
	from /opt/homebrew/Library/Homebrew/formula.rb:1326:in `block in brew'
	from /opt/homebrew/Library/Homebrew/formula.rb:2693:in `block (2 levels) in stage'
	from /opt/homebrew/Library/Homebrew/utils.rb:605:in `with_env'
	from /opt/homebrew/Library/Homebrew/formula.rb:2692:in `block in stage'
	from /opt/homebrew/Library/Homebrew/resource.rb:158:in `block (2 levels) in unpack'
	from /opt/homebrew/Library/Homebrew/download_strategy.rb:116:in `chdir'
	from /opt/homebrew/Library/Homebrew/download_strategy.rb:116:in `chdir'
	from /opt/homebrew/Library/Homebrew/download_strategy.rb:103:in `stage'
	from /opt/homebrew/Library/Homebrew/resource.rb:154:in `block in unpack'
	from /opt/homebrew/Library/Homebrew/mktemp.rb:77:in `block in run'
	from /opt/homebrew/Library/Homebrew/mktemp.rb:77:in `chdir'
	from /opt/homebrew/Library/Homebrew/mktemp.rb:77:in `run'
	from /opt/homebrew/Library/Homebrew/resource.rb:266:in `stage_resource'
	from /opt/homebrew/Library/Homebrew/resource.rb:153:in `unpack'
	from /opt/homebrew/Library/Homebrew/resource.rb:127:in `stage'
	from /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/forwardable.rb:230:in `stage'
	from /opt/homebrew/Library/Homebrew/formula.rb:2672:in `stage'
	from /opt/homebrew/Library/Homebrew/formula.rb:1319:in `brew'
	from /opt/homebrew/Library/Homebrew/build.rb:133:in `block in install'
	from /opt/homebrew/Library/Homebrew/utils.rb:605:in `with_env'
	from /opt/homebrew/Library/Homebrew/build.rb:128:in `install'
	from /opt/homebrew/Library/Homebrew/build.rb:229:in `<main>'

AlbrechtL, 2023-02-18:
I personally don't have MAC hardware. Does anyone else can help here?

AlbrechtL, 2024-10-08:
Closed because of inactivity.

closed 2024-10-08 by AlbrechtL