AlbrechtL/welle.io #791: Windows 10 X64, Welle.io Force Closes on starting program

starman316uk, 2023-08-08:
When starting program it force closes, I have tried different versions. Deleted the Welle.io registry entries and reinstalled

Zadig, installed Bulk in 0 and 1 drivers. 

It used to work fine and now it will not run. 

I uninstalled drivers, run program and program runs, states no device found, as soon as I install device, it force closes again

Will not run under Windows 11 too

andimik, 2023-08-08:
What about rtl_tcp or file?

starman316uk, 2023-08-08:
Fault bucket 2165765321850485869, type 5
Event Name: BEX64
Response: Not available
Cab Id: 0

Problem signature:
P1: welle-io.exe
P2: 0.0.0.0
P3: 63021645
P4: StackHash_02aa
P5: 0.0.0.0
P6: 00000000
P7: PCH_70_FROM_ntdll+0x000000000009DB34
P8: c0000005
P9: 0000000000000008
P10: 

Attached files:
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER5DA0.tmp.dmp
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER5FF3.tmp.WERInternalMetadata.xml
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER6023.tmp.xml
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER6030.tmp.csv
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER6060.tmp.txt

These files may be available here:
\\?\C:\ProgramData\Microsoft\Windows\WER\ReportArchive\AppCrash_welle-io.exe_57e4d2ff4e60b67e387144efff8766a587fd6af6_7f1b2543_ca353869-30b3-4ce2-87da-e70167241c9f

Analysis symbol: 
Rechecking for solution: 0
Report Id: 2f43c8a7-4e5b-402e-957d-0f55ef14f586
Report Status: 268435456
Hashed bucket: 1d2d9baf99c08aea2e0e5815e3c1fc6d
Cab Guid: 0

starman316uk, 2023-08-08:
Faulting application name: welle-io.exe, version: 0.0.0.0, time stamp: 0x63021645
Faulting module name: unknown, version: 0.0.0.0, time stamp: 0x00000000
Exception code: 0xc0000005
Fault offset: 0x00007ff696950000
Faulting process id: 0x17b0
Faulting application start time: 0x01d9c9d69b275331
Faulting application path: C:\Program Files\welle.io\welle-io.exe
Faulting module path: unknown
Report Id: 2f43c8a7-4e5b-402e-957d-0f55ef14f586
Faulting package full name: 
Faulting package-relative application ID:

starman316uk, 2023-08-08:
> What about rtl_tcp or file?

I tried replacing the rtlsdr.dll file and it still crashes when opening the program, program does not even display, it appears for a few seconds in the taskbar

starman316uk, 2023-08-08:
Managed to find another driver, from here 
https://www.ukwtv.de/cms/downloads-aside/281-dab-player-von-andreas-gsinn.html

Now works using their software, 

But not Welle.io, Now saying Error While opening device!

AlbrechtL, 2024-10-08:
Closed because of inactivity.

closed 2024-10-08 by AlbrechtL