AlbrechtL/welle.io #792: Looses signal exactly on the hour

DavidGoodenough, 2023-08-08:
I use Welle.io to listen to BBC Radio 3, 4 and WorldService.  Normally these work perfectly (thank you), but occasionally, and it is always exactly on the hour, it looses sync and what you get sounds like rather spiky static.

I am using 2.4 as packaged by Debian Sid.

Pressing Stop and then Play works every time to fix the problem.

Is there any way to get any debug information?

andimik, 2023-08-08:
BBC uses reconfiguration. This issue was described three years ago in https://github.com/AlbrechtL/welle.io/issues/587 but it is unsolved.

Therefore, please use another tool if possible, like https://github.com/KejPi/AbracaDABra

DavidGoodenough, 2023-09-30:
I tried anbracadbra but it fails to.load any of the national BBC channels

AlbrechtL, 2024-10-08:
Yes, duplicated with https://github.com/AlbrechtL/welle.io/issues/587.

closed 2024-10-08 by AlbrechtL