AlbrechtL/welle.io #793: welle-io does not scan for stations

mesoptafel, 2023-09-14:
system: debian 11 lxde minimal i386
I installed rtl-sdr and welle.io on this pc with synaptic. With rtl_test I can check that my cheap dvb-t dab dongle is recognized. Then I start welle-io and click on 'Start station scan'.  Nothing happens.
I tried a live iso (kali-linux) and did exactly the same thing. Now the 'Start station scan' completed and found some stations.

I think my minimal system is missing some dependencies.

alexmyczko, 2023-12-01:
can you try with debian 12?

mesoptafel, 2023-12-01:
I upgraded to debian 12 but the problem remains.
In the meantime I replaced the laptop for an system with debian 12 wayland kde. I also replaced the cheap dongle with the RTL-SDR V4. And this works
So my problem is no longer relevant.