AlbrechtL/welle.io #797: Scan is crashing with RTL-SDR V4 and the new drive

sorinpopa, 2024-01-11:
Scan is crashing with RTL-SDR V4 and the new drivers x86.
After crash, I have to remove the rtl-sdr dongle and plug it in again, then I can run welle.io and it works for some channels, but for others it's crashing again when I click play.
Maybe it's the new drivers...

andimik, 2024-01-12:
Is this an issue with welle.io or do you think it's worth informing the driver authors?

sorinpopa, 2024-01-12:
I would think it's worth informing the driver authors too.. Unless there is a a way (error log something) to find details about the crash.. 
Regards, Sorin

andimik, 2024-01-12:
What about using the driver together with rtl_tcp and send it to welle.io?

sorinpopa, 2024-01-12:
Hi, 
I apologize, I don't quite understand. Do you want me to send you the driver file? 
Regards, Sorin

andimik, 2024-01-12:
I mean, try to transmit the data to welle.io using rtl_tcp on port 1234.

sorinpopa, 2024-01-12:
Got it, thank you. I'll do that. 
Regards, Sorin

hansworzt, 2024-02-09:
Installed the RTL SDR Blog v4 dongle and it ran fine with SDR# following the quickstart guide.
However, the instructions to supposedly also make it work with Welle.IO do not work. It played one DAB+ station, and after clicking another it crashed and shut down - permanently. After restoring the replaced files that worked with my older RTL-SDR dongles, it worked again - but on the newest dongle nothing happens.

AlbrechtL, 2024-09-02:
Thanks for reporting. I updated the rtlsdr library. Can you try that latest developer version https://github.com/AlbrechtL/welle.io/actions/runs/10662651268/artifacts/1881000078, please.

AlbrechtL, 2024-10-08:
I even created a new unstable version. Can you try it again, please?
https://welle-io-nightlies.albrechtloh.de/20241008_ee183ef7_Windows_welle-io-setup_x64.exe

AlbrechtL, 2024-10-08:
It should be fixed in version 2.5. Please test again. For now, I'm closing this issue.