AlbrechtL/welle.io #799: output device null

Ale268, 2024-02-16:
Hello everyone

Thanks for the great software, sadly i cannot get it to run properly on my Raspberry 4.

It does not detect an audio output. The used soundcard is a hifiberry dac+ dsp connected via I2S.

The OS is Raspberry OS with ALSA for Audio management. 

Info:
    RaspiOS: 11.9 Bullseye 64-bit
    Linux kernel: 6.1.21 64-bit
    Pi model: Pi-4B 1.5 4GB
    MPD version: 0.23.14

[image]
[image]

Any Idea on how to fix this?

mpbraendli, 2024-02-19:
What's the output of `aplay -L` ?

Ale268, 2024-02-23:
Hello there I found the issue, the installation of pulseaudio fixed it.
Though, depending on the user it does work or not. 

If I start welle io as sudo it won't output whilst if I start it as PI it works.

closed 2024-02-25 by mpbraendli