AlbrechtL/welle.io #804: Does anyone know how to this program to work with new RTL V4 dongle

djpoole, 2024-03-21:
Does anyone know how to this program to work with new RTL V4 dongle 
[image]

sierra1964, 2024-03-21:
I have the same problem. He finds a lot of stations, but he only plays the respective station briefly and then there is silence. (interruption)

mpbraendli, 2024-03-21:
Have you tried with a manual gain setting?

sierra1964, 2024-03-21:
Unfortunately there is no setting option for this. At least I didn't find it. It finds a lot of channels (depending on the antenna) but no matter what you enter in the expert setting, it doesn't work correctly. As if the reception was too weak. I have at least 18 channels directly in my area. With SDR V3 or V5 it works for me.

AlbrechtL, 2024-04-20:
It seems that the RTL V4 creates some trouble.

Here is another non-working report: https://github.com/AlbrechtL/welle.io/issues/797
But here is an working report: https://github.com/AlbrechtL/welle.io/issues/793

rkarikari, 2024-06-29:
[image]
[image]
[image]
[image]

The snapshots above are from my android device.

Works perfectly for me.

The key is to go to .....
 "expert  settings"  and enable "expert mode". 

*** Make sure "Enable course corrector" is active. ***

I only have the Blog V4 dongles and they all work perfectly!
Have it working on all devices .... macbook, pc, android.

In sdrAngel, there's stuttering so I prefer to use welle.io

AlbrechtL, 2024-09-02:
Thanks for reporting. I updated the rtlsdr library. Can you try that latest developer version https://github.com/AlbrechtL/welle.io/actions/runs/10662651268/artifacts/1881000078, please.

rkarikari, 2024-09-02:
[image]
[image]

Deleted the old version I had and installed latest dev. version.
working well for me ..

rkarikari, 2024-09-02:
fyi .. the android version dated 20240830 gives the following error on android13
[image]

andimik, 2024-09-02:
Strange, I even can't install this version on Android 9

AlbrechtL, 2024-09-03:
> Deleted the old version I had and installed latest dev. version.
> working well for me ..

Thanks for the feedback

@rkarikari, @andimik Regarding the Android version open a new issue please.

AlbrechtL, 2024-09-03:
The new RTLSDR Windows driver will be included in the upcoming 2.5 release.

closed 2024-09-03 by AlbrechtL

AlbrechtL, 2024-10-08:
Unfortunately, have had to replace the rtl-sdr driver again because of this issue https://github.com/rtlsdrblog/rtl-sdr-blog/issues/56.

In this https://welle-io-nightlies.albrechtloh.de/20241008_ee183ef7_Windows_welle-io-setup_x64.exe unstable build the latest original rtlsdr driver is included. Can you test again, please?