AlbrechtL/welle.io #805: Airspy Mini on mãos not recognized

karterilla, 2024-04-06:
Hello and thanks for the software.
I'm trying to use an Airspy Mini on a macOS running latest OS Sonoma 14.4. It always saying, after installing the macport with command line and the rest of required packages, that "No valid devices found use Null device instead.
The Airspy Mini has DevID of 0x60a1 and VendID od 0x1d50.

[image]
Thanks

karterilla, 2024-04-06:
I do also try to use command line such as:
brew tap AlbrechtL/welle_io https://github.com/AlbrechtL/welle.io: no errors or output
while
brew install AlbrechtL/welle_io/welle.io
on make install gives this error back:
[ 85%] Built target welle-io
make: *** [all] Error 2

If reporting this issue please do so to (not Homebrew/brew or Homebrew/homebrew-core):
  albrechtl/welle_io
usr/local/Homebrew/Library/Homebrew/utils/github/api.rb:378:in `raise_error': Validation Failed: [{"message"=>"The listed users and repositories cannot be searched either because the resources do not exist or you do not have permission to view them.", "resource"=>"Search", "field"=>"q", "code"=>"invalid"}] (GitHub::API::ValidationFailedError)

\I will keep investigating.

AlbrechtL, 2024-10-08:
Sorry, I can't help you because I don't own an Apple computer.

closed 2024-10-08 by AlbrechtL