AlbrechtL/welle.io #808: FreeBSD 14.0: Can't scan for stations

ehaupt, 2024-05-05:
On FreeBSD `14.0-RELEASE`, I can't scan for radio stations with my `Realtek RTL2838UHIDIR`.

I am using the version from ports (version 2.4, pinned to git commit 0aacf578d77d79212483cd6fce436fb66523f63f). When starting up welle.io, I get:

```sh
2024-05-05T12:22:38.569 Debug: main: Set language "C"
2024-05-05T12:22:38.570 Debug: main: Error while loading language "C" use untranslated text (ie. English)
2024-05-05T12:22:38.731 Debug: Audio: Current sound output "default"
2024-05-05T12:22:38.901 Debug: Used input device: 0
2024-05-05T12:22:38.902 Debug: main: Set language "C"
2024-05-05T12:22:38.902 Debug: main: Error while loading language "C" use untranslated text (ie. English)
2024-05-05T12:22:38.903 Warning: qrc:/QML/settingpages/ExpertSettings.qml:34:1: QML ExpertSettings: Binding loop detected for property "implicitHeight"
2024-05-05T12:22:38.923 Debug: Used input device: 2
2024-05-05T12:22:38.923 Debug: Apply settings initially
2024-05-05T12:22:38.923 Debug: RadioController: Close device
2024-05-05T12:22:38.923 Info: InputFactory:Input device:auto
2024-05-05T12:22:38.924 Info: RTL_SDR: Open rtl-sdr
2024-05-05T12:22:38.924 Info: RTL_SDR: Found 1 devices. Uses the first working one
Found Rafael Micro R820T tuner
2024-05-05T12:22:39.370 Info: RTL_SDR:  Opening rtl-sdr device0
[R82XX] PLL not locked!
2024-05-05T12:22:39.486 Info: RTL_SDR:  Supported gain values29
2024-05-05T12:22:39.487 Info: RTL_SDR:  gain 49.6
2024-05-05T12:22:39.487 Info: RTL_SDR:  gain 48
2024-05-05T12:22:39.487 Info: RTL_SDR:  gain 44.5
2024-05-05T12:22:39.487 Info: RTL_SDR:  gain 43.9
2024-05-05T12:22:39.487 Info: RTL_SDR:  gain 43.4
2024-05-05T12:22:39.487 Info: RTL_SDR:  gain 42.1
2024-05-05T12:22:39.487 Info: RTL_SDR:  gain 40.2
2024-05-05T12:22:39.487 Info: RTL_SDR:  gain 38.6
2024-05-05T12:22:39.487 Info: RTL_SDR:  gain 37.2
2024-05-05T12:22:39.487 Info: RTL_SDR:  gain 36.4
2024-05-05T12:22:39.487 Info: RTL_SDR:  gain 33.8
2024-05-05T12:22:39.487 Info: RTL_SDR:  gain 32.8
2024-05-05T12:22:39.487 Info: RTL_SDR:  gain 29.7
2024-05-05T12:22:39.487 Info: RTL_SDR:  gain 28
2024-05-05T12:22:39.487 Info: RTL_SDR:  gain 25.4
2024-05-05T12:22:39.487 Info: RTL_SDR:  gain 22.9
2024-05-05T12:22:39.487 Info: RTL_SDR:  gain 20.7
2024-05-05T12:22:39.487 Info: RTL_SDR:  gain 19.7
2024-05-05T12:22:39.487 Info: RTL_SDR:  gain 16.6
2024-05-05T12:22:39.487 Info: RTL_SDR:  gain 15.7
2024-05-05T12:22:39.487 Info: RTL_SDR:  gain 14.4
2024-05-05T12:22:39.487 Info: RTL_SDR:  gain 12.5
2024-05-05T12:22:39.487 Info: RTL_SDR:  gain 8.7
2024-05-05T12:22:39.487 Info: RTL_SDR:  gain 7.7
2024-05-05T12:22:39.487 Info: RTL_SDR:  gain 3.7
2024-05-05T12:22:39.487 Info: RTL_SDR:  gain 2.7
2024-05-05T12:22:39.487 Info: RTL_SDR:  gain 1.4
2024-05-05T12:22:39.487 Info: RTL_SDR:  gain 0.9
2024-05-05T12:22:39.487 Info: RTL_SDR:  gain 0
2024-05-05T12:22:39.512 Debug: RadioController: AGC on
2024-05-05T12:22:39.512 Debug: Audio: Volume 1
2024-05-05T12:22:39.534 Info: RTL_SDR: Set bias tee to 0
2024-05-05T12:22:39.560 Warning: qrc:/QML/MainView.qml:154:13: QML ToolButton: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
2024-05-05T12:22:39.560 Warning: qrc:/QML/MainView.qml:251:13: QML ToolButton: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
2024-05-05T12:22:39.560 Warning: file:///usr/local/lib/qt5/qml/QtQuick/Controls.2/ApplicationWindow.qml: Conflicting properties 'visible' and 'visibility' for Window 'window'
2024-05-05T12:22:39.563 Debug: os: unix
2024-05-05T12:22:39.563 Debug: desktopAvailableWidth: 5120
2024-05-05T12:22:39.563 Debug: desktopAvailableHeight: 1440
2024-05-05T12:22:39.563 Debug: orientation: 2
2024-05-05T12:22:39.563 Debug: devicePixelRatio: 1
2024-05-05T12:22:39.563 Debug: pixelDensity: 4.283325370281892
2024-05-05T12:22:39.563 Debug: Creating component: qrc:/QML/RadioView.qml
2024-05-05T12:22:39.577 Debug: Creating component: qrc:/QML/MotView.qml
2024-05-05T12:22:39.582 Warning: QQmlComponent: Created graphical object was not placed in the graphics scene.
2024-05-05T12:23:14.159 Debug: main: Application closed
2024-05-05T12:23:14.187 Debug: GUI: Deleting CGUIHelper
2024-05-05T12:23:14.187 Debug: RadioController: Close device
2024-05-05T12:23:14.222 Debug: RadioController: Deleting CRadioController
```

According to your excellent tutorial video, I should have a three-dot icon to the right of the 'All stations' drop-down menu. As you can see in my snapshot, it does not appear.

[image]

Permissions to the device should not be a problem. When running dump1090 as the same user, I get:

```sh
$ dump1090
Found 1 device(s):
0: Realtek, RTL2838UHIDIR, SN: 00000001 (currently selected)
Found Rafael Micro R820T tuner
Max available gain is: 49.60
Setting gain to: 49.60
Exact sample rate is: 2000000.052982 Hz
Gain reported by device: 49.60
*8daa6c50ea1788669b3c08475997;
CRC: 475997 (ok)
DF 17: ADS-B message.
  Capability     : 5 (Level 2+3+4 (DF0,4,5,11,20,21,24,code7 - is on airborne))
  ICAO Address   : aa6c50
  Extended Squitter  Type: 29
  Extended Squitter  Sub : 2
  Extended Squitter  Name: Unknown
    Unrecognized ME type: 29 subtype: 2

*8daa6c505837839e5bdf359898b1;
CRC: 9898b1 (ok)
Single bit error fixed, bit 15
DF 17: ADS-B message.
  Capability     : 5 (Level 2+3+4 (DF0,4,5,11,20,21,24,code7 - is on airborne))
  ICAO Address   : aa6c50
  Extended Squitter  Type: 11
  Extended Squitter  Sub : 0
  Extended Squitter  Name: Airborne Position (Baro Altitude)
    F flag   : even
    T flag   : non-UTC
    Altitude : 10000 feet
    Latitude : 118573 (not decoded)
    Longitude: 122677 (not decoded)

*02a186b88450d3;
CRC: 8450d3 (ok)
DF 0: Short Air-Air Surveillance.
  Altitude       : 10000 feet
  ICAO Address   : aa6c50

*02a186b88450d3;
CRC: 8450d3 (ok)
DF 0: Short Air-Air Surveillance.
  Altitude       : 10000 feet
  ICAO Address   : aa6c50

*02a186b88450d3;
CRC: 8450d3 (ok)
DF 0: Short Air-Air Surveillance.
  Altitude       : 10000 feet
  ICAO Address   : aa6c50
...
```

Any help with getting welle.io to work on FreeBSD 14.0 would be much appreciated.

andimik, 2024-05-05:
Have you tried to use rtl_tcp?

andimik, 2024-05-06:
So the issue is solved and you will use rtl_tcp as it works now?

ehaupt, 2024-05-06:
Not quite :-) The original report was about the fact that welle.io does not show the three dot channel scan icon.

andimik, 2024-05-06:
I still can't follow you. Do you see the dots when you are using `rtl_tcp`?

ehaupt, 2024-05-06:
It seems like you'd like me to start rtl_tcp and then welle-io. I wasn't quite sure before.

However, doing this doesn't seem to change anything. As you can see in my original screenshot of welle.io, there aren't any three dot icons to initiate a scan.

andimik, 2024-08-04:
https://github.com/AlbrechtL/welle.io/issues/806

@ehaupt 

This seems to be a similar issue. The icons were not visible.

ehaupt, 2024-08-05:
> #806
> 
> @ehaupt
> 
> This seems to be a similar issue. The icons were not visible.

Confirmed. Clicking the area where it's **supposed** to be will open the context menu:

[image]

andimik, 2024-08-05:
So please try various (manual) gain settings and scan again. If not found, scan manually on the channels where your local signals are.

AlbrechtL, 2024-10-08:
Can you try version 2.5? In the meantime I'm closing this issue.

closed 2024-10-08 by AlbrechtL