AlbrechtL/welle.io #809: [rtl_tcp] How to enter a big (>8000) port number

HilmarSt, 2024-05-07:
I'm trying to use welle-io.exe (Windows version 2.4) with rtl_tcp. Due to proxy reasons, I have to enter a port with the number 8444 into the rtl_tcp configuration. With the wheel input (starting with port 1234) it takes ages to scroll to it.
In the past there were some command line options - but they seemed to be removed.
Is there an easy way to enter such a high port number?
May be you just state the Windows registry path where the port number is stored.

andimik, 2024-08-18:
#727 has a similar problem.

AlbrechtL, 2024-08-30:
Currently only 4 digit port numbers are possible. Sorry for that!

Theoretically you can compile welle.io by yourself and hard code the port number.

HilmarSt, 2024-09-12:
The digit count is not the problem.
The problem is how to get there:
"With the wheel input (starting with port 1234) it takes ages to scroll to 8444."

AlbrechtL, 2024-10-08:
I agree that we have something to change here. Let's move the discussion to this issue https://github.com/AlbrechtL/welle.io/issues/727.

closed 2024-10-08 by AlbrechtL

AlbrechtL, 2024-10-09:
In https://github.com/AlbrechtL/welle.io/commit/466be3df9894ca9f2b4a075902f2b16f42d4ca62 I changed to rtltcp settings (see screenshot). Now you can easily use big port numbers.

[image]

Feel free to test the latest unstable build: https://welle-io-nightlies.albrechtloh.de/20241009_466be3df_Windows_welle-io-setup_x64.exe