AlbrechtL/welle.io #810: Extreme badly Reception with DVB T stick and active antenna

Hammer20l, 2024-06-05:
Hallo Github,
I have a problem with my Welle.IO Setup. I use a DVB T Stick (MSI Digivox mini deluxe) which works on my notebook. I tryed 3 antennas, the one for all sv9323, a passive antenna from kabeldirekt and a no name mini antenna. With no antenna i get the channels in what i get with my mini radio, i only can hear a few seconds of sound with the kabeldirekt, no picture and only the channels what i get with full quality on my little radio.
Can anyone give me tips to solve the problem and get a normal antenne range?

andimik, 2024-06-05:
You need an antenna covering the VHF III Band, which is 174 - 230 MHz.

Hammer20l, 2024-06-05:
The Antennas are all compatible.

andimik, 2024-06-06:
Then try to use manual gain settings.

andimik, 2024-08-04:
@Hammer20l

Can you close this issue if solved?

AlbrechtL, 2024-10-08:
Closed because of inactivity.

closed 2024-10-08 by AlbrechtL