AlbrechtL/welle.io #811: No channels founds during scan with RTL-SDR-V4, but scanning process appears to happen

icooper9000, 2024-06-14:
Is there a work around for this? Is it a driver issue? I have DAB working on SDRangel which verifies the  antenna and connections that I have etc, but I would prefer a dedicated , simpler interface like welle.io. Does anybody know how to make this work?

rkarikari, 2024-06-29:
[image]
[image]
[image]
[image]

The snapshots above are from my android device.

Works perfectly for me.

The key is to go to .....
 "expert  settings"  and enable "expert mode". 

*** Make sure "Enable course corrector" is active. ***

I only have the Blog V4 dongles and they all work perfectly!
Have it working on all devices .... macbook, pc, android.

In sdrAngel, there's stuttering so I prefer to use welle.io

icooper9000, 2024-07-04:
Thanks for you message. 
I tried android after reading your response and I can get it to work on Android, no problem there, but I cannot get it to work on Window PC with same version but windows build.  I have also tried replacing rtlsdr.dll as recommended on rtlsdr blog, but that just caused it to crash.  
I tried using Abraca DAB app on windows, it works very well and there are no issues I can see.  I also experienced some stuttering on SDR Angel .

RadegasTCZ, 2024-08-04:
I think it might not be properly selecting the frequency bands because I have literally the same spectrum from 5A to 11A so there is something wrong in welle. RTL-SDR v4 dongle

[image]

[image]

This is 12D as an example in welle

[image]

And this is on SDR++, same gain as welle.

[image]

rkarikari, 2024-08-04:
All my rtlsdr BlogV4 dongles work well with welle.io.

os's : macos, pc, android, linux.... all work well.

It.s always an issue with scanning for a lock.

check the "coarse corrector" as shown in the pic in my post above.

RadegasTCZ, 2024-08-04:
it was enabled in that state by default and changing it literally changes absolutely nothing.

rkarikari, 2024-08-04:
I went through the same issues as you are.  I remember the frustration.

I found out the following...

The graph you posted is what welle.io shows when there is an issue with reception .... 
eg. inadequate  ( signal strength / signal-to-noise-ratio ) issues , poor antenna/location issues etc.

Digital radio requires a much higher signal quality than analogue.
example. mine do not work downstairs ... only upstairs.... although the graph looks nice like in your sdr++ pic.

Once you can receive analogue FM well with your Blog v4, you obviously have the correct drivers installed for sdr++

In my welle.io folder (windows), I have a file ... 

"libusb-1.0.dll     29july2023   signedby AIRSPY.com" 

 I must have put/replaced  this file there seperately (got rtlsdr files from airspy?... can't remember) . 

I went through the problems you are facing..... I remember the frustration.
but once I understood what the issue was,  I got it working everywhere.

Hope you find a solution

rkarikari, 2024-08-04:
The host device performance can also be an issue... so try with a more powerful  PC/phone/tablet.

RadegasTCZ, 2024-08-04:
I definitely suspect a software issue, I don't think Ryzen 7 7735HS is not good enough for this. Well yes I can see clearly the signal in SDR++ so it's definitely not reception issue. I was listening to 2m HAM band today as well 😃. I will have a look at the drivers tomorrow since it's getting late today.

rkarikari, 2024-08-04:
good luck tomorrow.  2m Ham band is mostly analogue. 

One thing I know for sure... If I was not fortunate to be working upstairs,  I would never have succeeded.
The signal looked great downstairs ... but it did not work there.
just something to take note of.
all the best.

andimik, 2024-08-04:
Have you tried `rtl_tcp` as input?

What about manual gain settings?

rkarikari, 2024-08-04:
> Have you tried `rtl_tcp` as input?
> 
> What about manual gain settings?

rtl_tcp was unreliable for me .
at low bitrates, it's barely ok . . i.e commercial FM, HAM digital modes etc. ( 300k and below)

2.0M    for DAB is a definite a no-go in my case.

Auto gain was easier to get working. 
Manual gain ( for me )  only works between 20 and 25.

when set below 20, I get your welle.io pics you posted above.

andimik, 2024-08-04:
No, `rtl_tcp` is very reliable for DAB. And it's definitely better when you want to use `welle-cli` as for me the signal is much more error-free.

rkarikari, 2024-08-04:
ok. good to know tcp works well for you.
for me, absolutely not!  wifi congestion issues. 
I always said "in my case".  

Obviously, whatever works for you is fine. 
All the best.

andimik, 2024-08-04:
Please note that tools like Qirx use `rtl_tcp` (or `rtl2_tcp`) and specific extensions from https://github.com/old-dab/rtlsdr and the performance is really good. It might decode ensembles which are not working in Welle.io.

Or have a look at AbracaDABra.

Both have a software related gain setting.

AlbrechtL, 2024-09-03:
@RadegasTCZ Can you test the latest unstable version, please? https://welle-io-nightlies.albrechtloh.de/20240903_985df6e7_Windows_welle-io-setup_x64.exe

RadegasTCZ, 2024-09-03:
> @RadegasTCZ Can you test the latest unstable version, please? https://welle-io-nightlies.albrechtloh.de/20240903_985df6e7_Windows_welle-io-setup_x64.exe

Hi there! Seems to be now tuning in.
Since it is unstable build, it does crash, and I don't seem to get any service working.

[image]

AlbrechtL, 2024-09-03:
Can you switch on the service details view in order to see the SNR?

Regarding the crashes: Currently I'm not able to reproduce it. So I have no idea where it happens :-(. Can you start welle.io with the argument `--log-file` and try to crash welle.io?
e.g. `welle-io.exe --log-file output_log.txt`

Please attach the log file here.

RadegasTCZ, 2024-09-03:
> Can you switch on the service details view in order to see the SNR?
>
> Regarding the crashes: Currently I'm not able to reproduce it. So I have no idea where it happens :-(. Can you start welle.io [...]

Well, it seems to be just disappearing, try using manual RF gain on the V4 dongle. I had two instances of it just disappearing randomly. I had a peek in the logs, and sadly, they just end with nothing. I will attach both of them.

[log_030924_1936_01.txt](https://github.com/user-attachments/files/16852766/log_030924_1936_01.txt)
[log_030924_1939_01.txt](https://github.com/user-attachments/files/16852767/log_030924_1939_01.txt)

As why it's not decoding, this is what is it complaining about.

[image]

AlbrechtL, 2024-09-03:
> As why it's not decoding, this is what is it complaining about.

Interesting. Can you enable the RAW file recorder and record e.g. 60 seconds.

How to:
1. Choose length e.g. 60 s
2. Click init
3. Wait 60 seconds
4. Click save
5. On your desktop you should have a file named `welle-io-record.iq`
6. Provide this file

Regarding the crashes I need more investigations.

AlbrechtL, 2024-09-04:
Regarding the crashes, I opened a new issue at rtlsdrblog: https://github.com/rtlsdrblog/rtl-sdr-blog/issues/56.

AlbrechtL, 2024-10-08:
Please try version 2.5 (https://github.com/AlbrechtL/welle.io/releases/tag/v2.5).

In the meantime, I'm closing this issue.

closed 2024-10-08 by AlbrechtL