AlbrechtL/welle.io #812: gain option has no effect

Hoerli1337, 2024-08-17:
Hi!
I've been playing around with the project for a week now.
I really like it so far.

I have an RTL-SDR v5 stick from nooelec.
This is connected to a Raspberry Pi 3B+.
It runs RaspberryOS 64bit as normal.

When trying out the best settings, I noticed that the gain parameter (-g) has absolutely no effect.
I live in a place where DAB+ is only possible with an external antenna and optimal positioning.
I wanted to fine-tune everything on the software side.
[image]

I can use manual amplification via the GUI.

Is this a bug or deliberately not possible?

andimik, 2024-08-18:
I have the same stick, 

```
rtl_test
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R860 or 820T/2 tuner
Supported gain values in dB (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6 
Supported bandwidth values in kHz: 290 375 420 470 600 860 950 1100 1200 1300 1503 1600 1753 1800 1953 2200 3000 5000 6000 7000 8000
Sampling at 2048000 S/s.
```

but if you use a too high manual gain for local channels, there is definitely a warning:

> RTL_SDR: ADC overload. Maybe you are using a too high gain.
> Info: ADC overload. Maybe you are using a too high gain.

But it seems that **a too high gain will set gain to 0**, as the value is "unknown"

see https://github.com/AlbrechtL/welle.io/blob/5481d57e728e03441c83b5586c2017f858f1dc31/src/input/rtl_sdr.cpp#L195-L197

I have tested with some gain settings:

# local mux (8 km away)

## gain 0.7

Ensemble ID | ECC | SNR | RX gain | Freq corr | Date | LTO | FIC CRC Errors | Tuned at | FCT0 frame received at
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
0xa304 | 0xe0 | 16.0 | 0.0 | 135 | 2024-8-18 7:46 UTC | 2 | 51 | 2024-08-18T07:46:15.616Z | 2024-08-18T07:46:24.700Z
---

## gain 9.7

Ensemble ID | ECC | SNR | RX gain | Freq corr | Date | LTO | FIC CRC Errors | Tuned at | FCT0 frame received at
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
0xa304 | 0xe0 | 11.0 | 15.7 | 150 | 2024-8-18 7:51 UTC | 2 | 233 | 2024-08-18T07:51:32.592Z | 2024-08-18T07:51:36.700Z
----

## gain 19.7 (overload, no lock)

Ensemble ID | ECC | SNR | RX gain | Freq corr | Date | LTO | FIC CRC Errors | Tuned at | FCT0 frame received at
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
0x0000 | 0x00 | 7.0 | 36.4 | -1 | 0-0-0 0:0 UTC | 0 | 276 | 2024-08-18T07:47:53.299Z | 2024-08-18T07:47:53.299Z
----

## gain 29.7 (will be set to 0, see SNR)

Ensemble ID | ECC | SNR | RX gain | Freq corr | Date | LTO | FIC CRC Errors | Tuned at | FCT0 frame received at
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
0xa304 | 0xe0 | 16.0 | 0.0 | 144 | 2024-8-18 7:49 UTC | 2 | 61 | 2024-08-18T07:49:33.463Z | 2024-08-18T07:49:48.699Z
----

## gain 39.7 (will be set to 0)

> RTL_SDR: Unknown gain count39

Ensemble ID | ECC | SNR | RX gain | Freq corr | Date | LTO | FIC CRC Errors | Tuned at | FCT0 frame received at
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
0xa304 | 0xe0 | 16.0 | 0.0 | 151 | 2024-8-18 7:54 UTC | 2 | 69 | 2024-08-18T07:53:56.875Z | 2024-08-18T07:54:00.700Z
----

## no manual gain 

Ensemble ID | ECC | SNR | RX gain | Freq corr | Date | LTO | FIC CRC Errors | Tuned at | FCT0 frame received at
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
0xa304 | 0xe0 | 21.0 | 1.4 | 148 | 2024-8-18 7:54 UTC | 2 | 69 | 2024-08-18T07:54:43.626Z | 2024-08-18T07:54:48.699Z
----

# weak mux (67 km away, amplifier needed)

## no manual gain 

Ensemble ID | ECC | SNR | RX gain | Freq corr | Date | LTO | FIC CRC Errors | Tuned at | FCT0 frame received at
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
0x9200 | 0xe4 | 17.0 | 25.4 | 181 | 2024-8-18 7:57 UTC | 2 | 267 | 2024-08-18T07:55:54.842Z | 2024-08-18T07:57:09.798Z
----

## gain 0.7 (no lock, signal is too weak)

Ensemble ID | ECC | SNR | RX gain | Freq corr | Date | LTO | FIC CRC Errors | Tuned at | FCT0 frame received at
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
0x0000 | 0x00 | 0.0 | 0.0 | 13018 | 0-0-0 0:0 UTC | 0 | 120 | 2024-08-18T07:57:50.973Z | 2024-08-18T07:57:50.973Z
---

## gain 9.7 (lock)

Ensemble ID | ECC | SNR | RX gain | Freq corr | Date | LTO | FIC CRC Errors | Tuned at | FCT0 frame received at
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
0x9200 | 0xe4 | 11.0 | 15.7 | 188 | 2024-8-18 7:58 UTC | 2 | 94 | 2024-08-18T07:58:43.253Z | 2024-08-18T07:58:45.798Z
---

## gain 19.7 (overload, error)

> RTL_SDR: ADC overload. Maybe you are using a too high gain.
> Info: ADC overload. Maybe you are using a too high gain.

Ensemble ID | ECC | SNR | RX gain | Freq corr | Date | LTO | FIC CRC Errors | Tuned at | FCT0 frame received at
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
0x9200 | 0xe4 | 15.0 | 36.4 | 191 | 2024-8-18 7:59 UTC | 2 | 92 | 2024-08-18T07:59:14.383Z | 2024-08-18T07:59:14.383Z
---

## gain 29.7, will be set to 0

** see the following message**

> RTL_SDR: Unknown gain count29

Ensemble ID | ECC | SNR | RX gain | Freq corr | Date | LTO | FIC CRC Errors | Tuned at | FCT0 frame received at
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
0x0000 | 0x00 | 0.0 | 0.0 | 0 | 0-0-0 0:0 UTC | 0 | 0 | 2024-08-18T08:00:15.373Z | 2024-08-18T08:00:15.373Z
----

# workaround

you can try with `rtl_tcp`. In a first console you type

```
rtl_tcp
```

and in a second console 

```
./welle-cli -c 8D -Dw 7979 -F rtl_tcp -g 29.7
```

> RTL_TCP_CLIENT: Tuner gain count: 29

BTW: the total amount of values is 29, but 29.7 is not the 29th value of course, see 

https://github.com/AlbrechtL/welle.io/blob/5481d57e728e03441c83b5586c2017f858f1dc31/src/input/rtl_tcp.h#L128-L130

As I am not a programmer I cannot tell you if also this code makes sense:

https://github.com/AlbrechtL/welle.io/blob/5481d57e728e03441c83b5586c2017f858f1dc31/src/input/rtl_sdr.cpp#L188-L191

Hoerli1337, 2024-08-18:
Thanks for the quick reply.
I have tested your commands, but they don't seem to work.

First I had to install rtl-sdr.
That wasn't on the Pi.

```
pi@raspberrypi3bplus:~ $ rtl_test
Found 1 device(s):
  0:  Nooelec, NESDR SMArt v5, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6
[R82XX] PLL not locked!
Sampling at 2048000 S/s.

Info: This tool will continuously read from the device, and report if
samples get lost. If you observe no further output, everything is fine.

Reading samples in async mode...
Allocating 15 zero-copy buffers
lost at least 68 bytes
```

I was then able to start rtl_tcp:
```
pi@raspberrypi3bplus:~ $ rtl_tcp
Found 1 device(s):
  0:  Nooelec, NESDR SMArt v5, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Tuned to 100000000 Hz.
listening...
Use the device argument 'rtl_tcp=127.0.0.1:1234' in OsmoSDR (gr-osmosdr) source
to receive samples in GRC and control rtl_tcp parameters (frequency, gain, ...).
client accepted! localhost 32796
Allocating 15 zero-copy buffers
set gain mode 1
set gain 0
set sample rate 2048000
set freq 201072000
```

welle-cli with different gain levels did not work.
According to the log, the gain remains at 29, regardless of what I enter via -g.
```
pi@raspberrypi3bplus:~ $ welle-cli -c 8D -w 7777 -F rtl_tcp -g 49.6
welle-cli 2.4+ds
InputFactory:Input device:rtl_tcp
OFDM-processor:restart
RTL_TCP_CLIENT: Try to connect to server 127.0.0.1:1234
Try to connect to: 127.0.0.1
RTL_TCP_CLIENT: Successful connected to server
RTL_TCP_CLIENT: Tuner type: 5 R820T
RTL_TCP_CLIENT: Tuner gain count: 29
ofdm-processor: SyncOnEndNull failed
ofdm-processor: Lost coarse sync (coarseCorrector: 0; fineCorrector: 0)
ofdm-processor: SyncOnPhase failed
Could not understand GET request /favicon.ico
```

Hoerli1337, 2024-08-18:
Okay, I'm stupid ...
I misunderstood the instructions on the net for the gain parameter.
Instead of the db values, the number must be specified.
I always specified the supported values.

After testing with the numbers, I realized that I can only receive anything at all at levels 0-3.
From level 4, nothing is received.

```
pi@raspberrypi3bplus:~ $ rtl_tcp
Found 1 device(s):
  0:  Nooelec, NESDR SMArt v5, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Tuned to 100000000 Hz.
listening...
Use the device argument 'rtl_tcp=127.0.0.1:1234' in OsmoSDR (gr-osmosdr) source
to receive samples in GRC and control rtl_tcp parameters (frequency, gain, ...).
client accepted! localhost 42526
Allocating 15 zero-copy buffers
set gain mode 1
set gain 0
set sample rate 2048000
set freq 201072000
set gain 14
set gain 27
set gain 37
set gain 77
set gain 87
set gain 125
set gain 144
set gain 157
set gain 166
set gain 197
set gain 207
set gain 229
set gain 254
set gain 280
set gain 297
set gain 328
set gain 338
set gain 328
set gain 297
```

AlbrechtL, 2024-12-26:
Closed because of inactivity.

closed 2024-12-26 by AlbrechtL