AlbrechtL/welle.io #814: apk cannot be installed under Android

andimik, 2024-09-03:
The Android version dated 20240830 gives the following error on Android 13 and also Android 9

[image]

First reported in https://github.com/AlbrechtL/welle.io/issues/804#issuecomment-2324405771 by user @rkarikari

AlbrechtL, 2024-09-03:
Does it also happens when you try the APKs from https://github.com/AlbrechtL/welle.io/actions/runs/10682830350?

rkarikari, 2024-09-03:
I get the very same error when attempting to install from the above link.

AlbrechtL, 2024-09-03:
Strange - something is going wrong during building ...

andimik, 2024-09-04:
Same here, the apk from the above link doesn't work neither.

AlbrechtL, 2024-09-30:
That could be the reason. @tenzap As you are the author of the Android Action pipeline can you update it and add the certificate process? Afterwards I can add my certificate.

kostyamat, 2024-10-17:
Hi, will this problem be solved?
And could you provide a link to the latest working APK?

tutacat, 2024-10-19:
You have to sign it yourself because they are not signing it. you may need to build it yourself.

rkarikari, 2024-10-25:
https://github.com/AlbrechtL/welle.io/actions/runs/11262592193/artifacts/2037144642

Tested ....  and it installed  and works great!
Thanks

> https://github.com/AlbrechtL/welle.io/actions/workflows/android.yml is the CI workflow that builds the android app, just select the most recent run and download the desired artifact (either any/all arch, or aarch64(arm64)-only).
> 
> It is actually being signed by the workflow now, so that can be included in releases binaries, and this issue is solved.

andimik, 2024-10-25:
Strange, the link works on my PC only. If I want to download it from my mobile, I get a 404 error from github ...

phenosys, 2025-01-19:
Hello,
I was interested in trying out the software on Android, but I am getting "This workflow was disabled manually. " error. Are you planning to reactivate building the software for Android in the near future?
(I am trying this logged in on Desktop browser).
Thanks! 
In the meantime I will try the Ubuntu version.

AlbrechtL, 2025-01-19:
> I was interested in trying out the software on Android, but I am getting "This workflow was disabled manually. " error. Are you planning to reactivate building the software for Android in the near future?

The Android workflow is unmaintained, currently. For that reason, it is disabled. It would be very nice if somebody would maintain it and providing PRs for it!

AlbrechtL, 2025-03-07:
@mi-hol Thanks for mentioning it! Can you create a pull request for the readme.md in the next branch, please?

leonam72, 2026-02-22:
Restores Android build using Qt 6.5.3 via aqtinstall  
Documents required modules and setup (BUILDING_ANDROID.md)  
Makes Qt6Quick3DUtils optional to avoid hard failures when not installed  

This restores Android build capability which was previously broken.

https://github.com/AlbrechtL/welle.io/pull/900

I enjoy reviving unmaintained parts of open-source projects — had some help from AI along the way. Hope this is useful 🙂