AlbrechtL/welle.io #815: Disable switching channel from webui

skinkie, 2024-09-09:
We would like to disable the channel switching from the webinterface, it should only use the provided channel from the command line argument of `welle-cli`. Is such functionality already implemented?

andimik, 2024-09-09:
Yes, already implemented, you just need to modify https://github.com/AlbrechtL/welle.io/blob/5481d57e728e03441c83b5586c2017f858f1dc31/src/welle-cli/index.js#L1

https://dabmonitor.com/test-1-windows.html

This was one project, but it is currently broken. Here, also switching channels was disabled.

skinkie, 2024-09-09:
I understand that this would give a _javascript_ option, but would it from API (security perspective) still possible?

skinkie, 2024-09-09:
Conceptually I want to prevent the retune. So some kind of access control on those features. Hence monitoring is public, but it must be at the same station.
https://github.com/AlbrechtL/welle.io/blob/5481d57e728e03441c83b5586c2017f858f1dc31/src/welle-cli/webradiointerface.cpp#L517

mpbraendli, 2024-10-09:
This is not implemented, we don't have time to implement this, but would welcome a patch.

closed 2024-10-09 by mpbraendli

skinkie, 2024-10-09:
> This is not implemented, we don't have time to implement this, but would welcome a patch.

Would you accept that the channel is fixed once it has been given as the command line argument, or is pre-tuning and later changing something that is conceptually supported.

mpbraendli, 2024-10-09:
Not sure I understand your question. I'd add a `--forbid-channel-change` option, that makes the backend refuse the POST to change channels.

Or if you think the idea further, have a `--public-instance` flag (or something like that), that makes all settings read-only. Ideally, the UI would also have the controls disabled.