AlbrechtL/welle.io #817: Windows welle.io has issues with sound breaking up

eth72419, 2024-09-21:
Hello,

I'm experiencing an issue with welle.io on Windows.

The sound keeps breaking up (eg. pauses and restarts) - not even the muddy bubbling sound. The antenna icon flashes on and off with constant RS errors and frequency correction. It's quite strange - this happens pretty much every 15-20 seconds, while there are no issues with this on the Android version.

It's not an issue with the signal - the RTL-SDR is connected straight to the a spare radio port using a coax cable and connector. I have also tried connecting an antenna straight to the port - with no avail.

This is likely not an issue with the RTL-SDR or computer itself, since Android works fine?

Thanks for your help - if there is any more info needed please reply!

eth72419, 2024-09-21:
I have tried using manual gain with no change in the issue

AlbrechtL, 2024-10-08:
Can you try the latest unstabled build please?
https://welle-io-nightlies.albrechtloh.de/20241008_ee183ef7_Windows_welle-io-setup_x64.exe

AlbrechtL, 2024-10-08:
It would be very nice if you could post some screenshots.

AlbrechtL, 2024-12-26:
Closed because of inactivity.

closed 2024-12-26 by AlbrechtL