AlbrechtL/welle.io #818: Building on Linux Mint (Ubuntu 24.04) fails due to QAudioSink

andimik, 2024-10-16:
I'm running Linux Mint which is based on Ubuntu 24.04.

Due to the fact that the AppImage does not work, I wanted to build it (both `master` and `next` branch) from sources (both with `qmake` and QtCreator), but it seem I needed to install some additional packages (which are not mentioned on your Readme, obviously), otherwise I get the following error:

```
make
cd src/welle-gui/ && ( test -e Makefile || /usr/lib/qt5/bin/qmake -o Makefile /home/andreas/apps/welle.io/src/welle-gui/welle-gui.pro ) && make -f Makefile 
Project MESSAGE: Current git hash = 6a1adf03
Project MESSAGE: BUILD_DATE = 1728505781
Project ERROR: Unknown module(s) in QT: quick qml charts multimedia
make: *** [Makefile:47: sub-src-welle-gui-make_first] Error 3
```

`sudo apt install qtdeclarative5-dev qtmultimedia5-dev libqt5charts5-dev`

But then I got the error:

```
In file included from ../../../src/welle-gui/radio_controller.h:45,
                 from ../../../src/welle-gui/main.cpp:44:
../../../src/welle-gui/audio_output.h:36:10: fatal error: QAudioSink: No such file or directory
   36 | #include <QAudioSink>
      |          ^~~~~~~~~~~~
compilation terminated.
make[1]: *** [Makefile:1256: main.o] Error 1
make[1]: Leaving directory '/home/andreas/apps/welle.io/build/src/welle-gui'
make: *** [Makefile:47: sub-src-welle-gui-make_first] Error 2
```
see https://doc.qt.io/qt-6/qaudiosink.html

The funny thing is, that this line is from 2022 (see https://github.com/AlbrechtL/welle.io/blame/a910d9d2f8116aa84d476eb39efa1c24ccb4b429/src/welle-gui/audio_output.h#L36 ) and I remember  I could build it on my previous Ubuntu installation.

Then I checked if there is a file already on my system:

```
andreas@E535:/usr/include/x86_64-linux-gnu/qt6/QtMultimedia$ ls -l | grep QAudioSink
-rw-r--r-- 1 root root   24 Dec 12  2022 QAudioSink
```

If you need that:

```
qt6-multimedia-dev is already the newest version (6.4.2-11build3).
```

andimik, 2024-10-16:
BTW: I could build it with cmake, but then the app is only visible as icon in the start menu.

```
andreas@E535:~/apps/welle.io/build$ make
[  1%] Built target welle-io_qmlimportscan
[  1%] Built target welle-io_autogen_timestamp_deps
[  3%] Automatic MOC for target welle-io
[  3%] Built target welle-io_autogen
[  5%] Generating qrc_resources.cpp
[  7%] Building CXX object CMakeFiles/welle-io.dir/welle-io_autogen/mocs_compilation.cpp.o
[  9%] Building CXX object CMakeFiles/welle-io.dir/src/welle-gui/main.cpp.o
[ 11%] Building CXX object CMakeFiles/welle-io.dir/src/welle-gui/audio_output.cpp.o
[ 12%] Building CXX object CMakeFiles/welle-io.dir/src/welle-gui/mot_image_provider.cpp.o
[ 14%] Building CXX object CMakeFiles/welle-io.dir/src/welle-gui/gui_helper.cpp.o
[ 16%] Building CXX object CMakeFiles/welle-io.dir/src/welle-gui/radio_controller.cpp.o
/home/andreas/apps/welle.io/src/welle-gui/radio_controller.cpp: In member function 'void CRadioController::initialise()':
/home/andreas/apps/welle.io/src/welle-gui/radio_controller.cpp:609:21: warning: loop variable 'param_value' creates a copy from type 'const std::pair<const DeviceParam, std::__cxx11::basic_string<char> >' [-Wrange-loop-construct]
  609 |     for (const auto param_value : deviceParametersString) {
      |                     ^~~~~~~~~~~
/home/andreas/apps/welle.io/src/welle-gui/radio_controller.cpp:609:21: note: use reference type to prevent copying
  609 |     for (const auto param_value : deviceParametersString) {
      |                     ^~~~~~~~~~~
      |                     &
[ 18%] Building CXX object CMakeFiles/welle-io.dir/src/welle-gui/debug_output.cpp.o
[ 20%] Building CXX object CMakeFiles/welle-io.dir/src/welle-gui/waterfallitem.cpp.o
[ 22%] Building CXX object CMakeFiles/welle-io.dir/src/welle-gui/mpris/mpris.cpp.o
[ 24%] Building CXX object CMakeFiles/welle-io.dir/src/welle-gui/mpris/mpris_mp2.cpp.o
[ 25%] Building CXX object CMakeFiles/welle-io.dir/src/welle-gui/mpris/mpris_mp2_player.cpp.o
[ 27%] Building CXX object CMakeFiles/welle-io.dir/qrc_resources.cpp.o
[ 29%] Building CXX object CMakeFiles/welle-io.dir/src/backend/dab-audio.cpp.o
[ 31%] Building CXX object CMakeFiles/welle-io.dir/src/backend/decoder_adapter.cpp.o
[ 33%] Building CXX object CMakeFiles/welle-io.dir/src/backend/dab_decoder.cpp.o
[ 35%] Building CXX object CMakeFiles/welle-io.dir/src/backend/dabplus_decoder.cpp.o
[ 37%] Building CXX object CMakeFiles/welle-io.dir/src/backend/charsets.cpp.o
[ 38%] Building CXX object CMakeFiles/welle-io.dir/src/backend/dab-constants.cpp.o
[ 40%] Building CXX object CMakeFiles/welle-io.dir/src/backend/mot_manager.cpp.o
[ 42%] Building CXX object CMakeFiles/welle-io.dir/src/backend/pad_decoder.cpp.o
[ 44%] Building CXX object CMakeFiles/welle-io.dir/src/backend/eep-protection.cpp.o
[ 46%] Building CXX object CMakeFiles/welle-io.dir/src/backend/fib-processor.cpp.o
/home/andreas/apps/welle.io/src/backend/fib-processor.cpp: In member function 'int16_t FIBProcessor::HandleFIG0Extension8(uint8_t*, int16_t, uint8_t)':
/home/andreas/apps/welle.io/src/backend/fib-processor.cpp:459:17: warning: unused variable 'SCid' [-Wunused-variable]
  459 |         int16_t SCid = getBits(d, lOffset + 4, 12);
      |                 ^~~~
/home/andreas/apps/welle.io/src/backend/fib-processor.cpp:466:17: warning: unused variable 'SubChId' [-Wunused-variable]
  466 |         int16_t SubChId = getBits_6(d, lOffset + 4);
      |                 ^~~~~~~
[ 48%] Building CXX object CMakeFiles/welle-io.dir/src/backend/fic-handler.cpp.o
[ 50%] Building CXX object CMakeFiles/welle-io.dir/src/backend/msc-handler.cpp.o
[ 51%] Building CXX object CMakeFiles/welle-io.dir/src/backend/freq-interleaver.cpp.o
[ 53%] Building CXX object CMakeFiles/welle-io.dir/src/backend/ofdm-decoder.cpp.o
[ 55%] Building CXX object CMakeFiles/welle-io.dir/src/backend/ofdm-processor.cpp.o
[ 57%] Building CXX object CMakeFiles/welle-io.dir/src/backend/phasereference.cpp.o
[ 59%] Building CXX object CMakeFiles/welle-io.dir/src/backend/phasetable.cpp.o
[ 61%] Building CXX object CMakeFiles/welle-io.dir/src/backend/tii-decoder.cpp.o
[ 62%] Building CXX object CMakeFiles/welle-io.dir/src/backend/protTables.cpp.o
[ 64%] Building CXX object CMakeFiles/welle-io.dir/src/backend/radio-receiver.cpp.o
[ 66%] Building CXX object CMakeFiles/welle-io.dir/src/backend/tools.cpp.o
[ 68%] Building CXX object CMakeFiles/welle-io.dir/src/backend/uep-protection.cpp.o
[ 70%] Building CXX object CMakeFiles/welle-io.dir/src/backend/viterbi.cpp.o
[ 72%] Building CXX object CMakeFiles/welle-io.dir/src/various/Socket.cpp.o
[ 74%] Building CXX object CMakeFiles/welle-io.dir/src/various/Xtan2.cpp.o
[ 75%] Building CXX object CMakeFiles/welle-io.dir/src/various/channels.cpp.o
/home/andreas/apps/welle.io/src/various/channels.cpp: In member function 'std::string Channels::getChannelForFrequency(int)':
/home/andreas/apps/welle.io/src/various/channels.cpp:229:21: warning: loop variable 'c_f' creates a copy from type 'const std::pair<const std::__cxx11::basic_string<char>, int>' [-Wrange-loop-construct]
  229 |     for (const auto c_f : frequencyMap) {
      |                     ^~~
/home/andreas/apps/welle.io/src/various/channels.cpp:229:21: note: use reference type to prevent copying
  229 |     for (const auto c_f : frequencyMap) {
      |                     ^~~
      |                     &
[... 5 lines cut ...]
[ 87%] Building C object CMakeFiles/welle-io.dir/src/libs/fec/init_rs_char.c.o
[... cut ...]
2024-10-17T00:49:33.010 Debug: main: Set language "C"
2024-10-17T00:49:33.010 Debug: main: Error while loading language "C" use untranslated text (ie. English)
2024-10-17T00:49:33.051 Debug: Audio: Current sound output "Built-in Audio Analog Stereo"
2024-10-17T00:49:33.274 Warning: QQmlApplicationEngine failed to load component
2024-10-17T00:49:33.274 Warning: qrc:/QML/MainView.qml:106:5: Settings is not a type
```

[CMakeCache.txt](https://github.com/user-attachments/files/17403503/CMakeCache.txt)

AlbrechtL, 2024-10-29:
welle.io based on Qt6 now. See https://github.com/AlbrechtL/welle.io?tab=readme-ov-file#debian--ubuntu-linux for details.

You need to install the following libraries 
`sudo apt install libqt6charts6-dev qt6-base-dev qt6-tools-dev-tools qt6-multimedia-dev qt6-wayland-dev libqt6core5compat6-dev libqt63dquick6 libqt6qml6 qt6-declarative-dev  qml6-module-qtquick-controls qml6-module-qtquick qml6-module-qt5compat-graphicaleffects qml6-module-qtcharts  qml6-module-qtmultimedia qml6-module-qtquick-window qml6-module-qtquick-layouts qml6-module-qtqml-workerscript qml6-module-qtwayland-compositor qml6-module-qtquick-templates qtcreator`

AlbrechtL, 2024-12-13:
Closed due inactivity.

closed 2024-12-13 by AlbrechtL

andimik, 2025-03-05:
No, this does not work. I tried to install all the packages you have mentioned, but the error is still there.

andimik, 2025-03-05:
I found this on https://github.com/c3d/db48x/issues/1288

Maybe this helps?

```
apt-file search "QAudioSink"
qt6-multimedia-dev: /usr/include/x86_64-linux-gnu/qt6/QtMultimedia/QAudioSink
```