AlbrechtL/welle.io #823: Missing qml6-module-qtcore and qml6-module-qtquick-dialogs in Readme

andimik, 2024-11-10:
https://github.com/AlbrechtL/welle.io/blob/654e45761095a4777597abdbb35a31718214c92b/README.md?plain=1#L149

There are two package missing in this list: `sudo apt-get install qml6-module-qtcore qml6-module-qtquick-dialogs` are necessary, too. 

Otherwise you will **not see anything** after starting `welle-io`.

```
./welle-io
2024-11-10T12:15:56.547 Debug: main: Platform name "wayland"
2024-11-10T12:15:56.547 Debug: main: Set language "de_DE"
2024-11-10T12:15:56.560 Debug: Audio: Current sound output "Internes Audio Analog Stereo"
2024-11-10T12:15:56.592 Warning: QQmlApplicationEngine failed to load component
2024-11-10T12:15:56.592 Warning: qrc:/QML/MainView.qml:32:1: module "QtCore" is not installed
```

And, if you don't install `qml6-module-qtquick-dialogs` you will get the following error:

```
2024-11-10T12:19:44.547 Debug: Error:qrc:/QML/MotView.qml:29 module "QtQuick.Dialogs" is not installed
```

AlbrechtL, 2024-12-26:
Which Linux distribution do you are using?

andimik, 2024-12-26:
I have several laptops, I can't remember, what I was using some weeks ago.

But I have Ubuntu 24.10 and 24.04 (in detail Linux Mint I guess).

AlbrechtL, 2024-12-27:
Ok. Can you send me a pull request please?

andimik, 2024-12-27:
Ok, done,

AlbrechtL, 2024-12-27:
Thanks for the pull request. I just merged it.

closed 2024-12-27 by AlbrechtL