AlbrechtL/welle.io #824: Same service in different ensembles just listed once

andimik, 2024-11-10:
# problem

In my region, some (identical) services can be received on different channels (here Radio Antena, Radio Bob and Toti MS on both muxes, R2W and R2E Slovenia). It's similar to https://github.com/AlbrechtL/welle.io/issues/819 , but this is for `welle-cli` only.

But **welle-io** will list them only once (the lower channel 12B) after a full scan. Unfortunately, it's the weaker channel ...

[image]

# expectation

Either both entries shall be listed or there should be an option to switch between the two channels (like in AbracaDABra).

# workaround

The only workaround is to delete the program list and scan manually, where you first begin with the strong muxes and then only scan the weaker ones.

[image]

Cmdr-Zod, 2024-12-22:
I am suffering from the same problem. I don't know how my traditional Receiver from TEAC is doing it, but I believe he is deciding based on the signal strength, which ensemble he uses for a certain station. I would be fine with a manual selection though.

AlbrechtL, 2024-12-26:
Yes, welle.io uses the first found stations and ignores if more stations with the same are found. This is regardless the channel strengths.

@andimik Can you post in https://github.com/AlbrechtL/welle.io/issues/487 a screenshot how AbracaDABra is doing it?

Closed because it is duplicated with https://github.com/AlbrechtL/welle.io/issues/487.

closed 2024-12-26 by AlbrechtL

AlbrechtL, 2024-12-28:
FYI: I'm proposing a possible solution here: https://github.com/AlbrechtL/welle.io/issues/487#issuecomment-2563988969