AlbrechtL/welle.io #825: Wrong DAB time

andimik, 2024-11-10:
I've recognized that the DAB time is wrong, it shows UTC-1 as GMT.

I'm living in Austria, so it is currently 12:46 local time (UTC +1h).

# Welle-io

[image]

Expected time should be 11:42, not 10:42.

# AbracaDABra

```
Ensemble:
0xA304 DAB+ MUX 2-StBK [ MUX2StBK ] ECC: 0xE0, UTC 60 min, INT: 1, alarm announcements: 0
```

[image]

Tested on 6 muxes from 2 countries, so it's not a broadcaster issue.

andimik, 2024-11-10:
I'm not (!) a programmer, but what about this?

```
diff --git a/src/backend/fib-processor.cpp b/src/backend/fib-processor.cpp
index 3cd18724..09f62830 100644
--- a/src/backend/fib-processor.cpp
+++ b/src/backend/fib-processor.cpp
@@ -516,7 +516,7 @@ void FIBProcessor::FIG0Extension10(uint8_t *fig)
     dateTime.year = Y;
     dateTime.month = M;
     dateTime.day = D;
-    dateTime.hour = getBits_5(fig, offset + 21);
+    dateTime.hour = getBits_5(fig, offset + 21) +1;
     if (getBits_6(fig, offset + 26) != dateTime.minutes)
         dateTime.seconds =  0;  // handle overflow
```

No idea if this solves it even for 23 hrs.

AlbrechtL, 2024-12-16:
Thanks for reporting!

Fixed in https://github.com/AlbrechtL/welle.io/commit/bd80a78bd164f3581d5ce88f0f2f7990c8174e2d.

closed 2024-12-16 by AlbrechtL