AlbrechtL/welle.io #829: Stationlist does not scroll, when using keyboard shortcuts

ErrorSource, 2024-12-10:
Hi there,

thank you very much for this great project. I am using welle.io combined with a music media server in my car. Works almost perfect. A USB scroll knob is used for controlling welle.io, sending keyboard shortcuts 'n' and 'p' for playing next and previous station in list. However, using the keyboard shortcuts does not scroll the station list, if the selected station goes beyond the actual visible station list on the left side.

Would it be possible to add the scrolling behaviour on going down the station list via keyboard shortcuts? The would be great.

ErrorSource, 2024-12-11:
Thanks for your quick answer!

Maybe there are no more keyboard shortcuts needed. If i use the existing keyboard shortcuts `n` and `p`, the station changes and the according entry in the station list is getting highlighted. But the station list will not "move"/scroll with the actual selected station, if it gets out of visible scope. I can not see, what's the next stations in list.

Normally you would scroll the station list with your mouse (or via touchscreen) and then choose a station. In my case, the station is changed via keyboard shortcuts and the station list does not scroll up or down in sync with actual selected station.

Like in your screenshot on welle.io:
[image]

Here "DLF" is selected as actual station. If you press `n`, station "DLF Kultur" would be tuned in and highlighted in station list, but you could only see half of it, because the list does not move. If you would press `n` again, the next station would be tuned, but you couldn't see which one at first sight.

Is it possible to let the list move up and down according to the selected station? If not, could you add keyboard shortcuts for changing the group of stations ("All stations", "Favorites", etc.)? Maybe `s` and `x`? But any other keys would be also great. Than I would group the stations to always fit into visible station list height.

Thanks in advance!

AlbrechtL, 2024-12-27:
Fixed in https://github.com/AlbrechtL/welle.io/commit/c99d2184dcaa1d155b72d8a1b92fbeac33dd982e.

Please test the unstable build at https://welle-io-nightlies.albrechtloh.de/20241227_c99d2184_Windows_welle-io-setup_x64.exe.

closed 2024-12-27 by AlbrechtL

AlbrechtL, 2025-01-03:
Great! Thanks for the feedback!