AlbrechtL/welle.io #845: [next] error: ‘UTC’ is not a member of ‘QTimeZone’

andimik, 2025-03-05:
I could not build the latest commit from next branch due to this following error (Ubuntu 24.04, Linux Mint)

```
make
[  1%] Built target welle-io_qmlimportscan
[  1%] Built target welle-io_autogen_timestamp_deps
[  3%] Built target welle-io_autogen
[  5%] Building CXX object CMakeFiles/welle-io.dir/src/welle-gui/radio_controller.cpp.o
/home/andreas/apps/welle.io/src/welle-gui/radio_controller.cpp: In member function ‘void CRadioController::displayDateTime(const dab_date_time_t&)’:
/home/andreas/apps/welle.io/src/welle-gui/radio_controller.cpp:894:44: error: ‘UTC’ is not a member of ‘QTimeZone’
  894 |     currentDateTime.setTimeZone(QTimeZone::UTC);
      |                                            ^~~
make[2]: *** [CMakeFiles/welle-io.dir/build.make:276: CMakeFiles/welle-io.dir/src/welle-gui/radio_controller.cpp.o] Fehler 1
make[1]: *** [CMakeFiles/Makefile2:89: CMakeFiles/welle-io.dir/all] Fehler 2
```

I'm not a programmer, but I tried some code I found on various pages, so this helped me by adding `()` and writing `utc` instead of `UTC`:

```
diff --git a/src/welle-gui/radio_controller.cpp b/src/welle-gui/radio_controller.cpp
index 9520e94d..f8031989 100644
--- a/src/welle-gui/radio_controller.cpp
+++ b/src/welle-gui/radio_controller.cpp
@@ -891,7 +891,7 @@ void CRadioController::displayDateTime(const dab_date_time_t& dateTime)
 
     // int OffsetFromUtc = dateTime.hourOffset * 3600 +
     //                     dateTime.minuteOffset * 60;
-    currentDateTime.setTimeZone(QTimeZone::UTC);
+    currentDateTime.setTimeZone(QTimeZone::utc());
 
     emit dateTimeChanged(currentDateTime);
 }
```

see https://doc.qt.io/qt-6.2/qtimezone.html

```
[ 87%] Linking CXX executable welle-io
[100%] Built target welle-io
```

But I cannot start the app right now, it's only an icon in the tray. Will check why.

AlbrechtL, 2025-03-07:
Which Qt version did you use? I tested the code with Qt 6.9.

andimik, 2025-03-08:
Same here. Please remove the `sudo apt-get install` commands from the Readme and mention the minimum Qt version which is required to build it.

closed 2025-03-08 by andimik

AlbrechtL, 2025-03-08:
To update the readme.md is in my todo list. Before I will release the version 2.7 I will update it. Sorry for the inconveniences!

evertverduin, 2025-03-09:
Very annoying, had the same issue