AlbrechtL/welle.io #846: Settings is not a type

andimik, 2025-03-05:
When building latest commit from next branch under Ubuntu 24.04 = Linux Mint, the app will only be start as icon.

The workaround is to comment out all lines with `Settings`:

```
git diff
diff --git a/src/welle-gui/QML/GeneralView.qml b/src/welle-gui/QML/GeneralView.qml
index 033f3d42..16048c5e 100644
--- a/src/welle-gui/QML/GeneralView.qml
+++ b/src/welle-gui/QML/GeneralView.qml
@@ -41,9 +41,9 @@ GridLayout {
     property bool isExpert: false
     property bool isPortrait: false
 
-    Settings {
-        property alias viewComponents: gridLayout.serialized
-    }
+//    Settings {
+//        property alias viewComponents: gridLayout.serialized
+//    }
 
     function addComponent(path, row, column) {
         // Check of component already exists
diff --git a/src/welle-gui/QML/MainView.qml b/src/welle-gui/QML/MainView.qml
index 1ca61550..9fecbd55 100644
--- a/src/welle-gui/QML/MainView.qml
+++ b/src/welle-gui/QML/MainView.qml
@@ -101,14 +101,14 @@ ApplicationWindow {
         isLoaded = true
     }
 
-    Settings {
-        property alias width : mainWindow.width
-        property alias height : mainWindow.height
-        property alias stationListSerialize: stationList.serialized
-        property alias favoritsListSerialize: favoritsList.serialized
-        property alias stationListBoxIndex: stationListBox.currentIndex
-        property alias volume: volumeSlider.value
-    }
+//    Settings {
+//        property alias width : mainWindow.width
+//        property alias height : mainWindow.height
+//        property alias stationListSerialize: stationList.serialized
+//        property alias favoritsListSerialize: favoritsList.serialized
+//        property alias stationListBoxIndex: stationListBox.currentIndex
+//        property alias volume: volumeSlider.value
+//    }
 
     header: ToolBar {
         id: overlayHeader
diff --git a/src/welle-gui/radio_controller.cpp b/src/welle-gui/radio_controller.cpp
index 9520e94d..f8031989 100644
--- a/src/welle-gui/radio_controller.cpp
+++ b/src/welle-gui/radio_controller.cpp
@@ -891,7 +891,7 @@ void CRadioController::displayDateTime(const dab_date_time_t& dateTime)
 
     // int OffsetFromUtc = dateTime.hourOffset * 3600 +
     //                     dateTime.minuteOffset * 60;
-    currentDateTime.setTimeZone(QTimeZone::UTC);
+    currentDateTime.setTimeZone(QTimeZone::utc());
 
     emit dateTimeChanged(currentDateTime);
 }
```

[image]

It will at least show the main window, but all menus are empty.

```
2025-03-05T20:26:44.724 Warning: qrc:/QML/settingpages/ExpertSettings.qml:40:5: Settings is not a type
2025-03-05T20:26:44.724 Warning: qrc:/QML/MainView.qml:833:13: TypeError: Cannot read property 'implicitHeight' of null
2025-03-05T20:26:44.724 Warning: qrc:/QML/MainView.qml:833:13: TypeError: Cannot read property 'implicitHeight' of null
2025-03-05T20:26:44.735 Warning: qrc:/QML/settingpages/GlobalSettings.qml:46:5: Settings is not a type
2025-03-05T20:26:44.735 Warning: qrc:/QML/MainView.qml:814:13: TypeError: Cannot read property 'implicitHeight' of null
2025-03-05T20:26:44.735 Warning: qrc:/QML/MainView.qml:814:13: TypeError: Cannot read property 'implicitHeight' of null
2025-03-05T20:26:44.738 Warning: qrc:/QML/settingpages/ChannelSettings.qml:40:5: Settings is not a type
2025-03-05T20:26:44.738 Warning: qrc:/QML/MainView.qml:797:13: TypeError: Cannot read property 'implicitHeight' of null
2025-03-05T20:26:44.738 Warning: qrc:/QML/MainView.qml:797:13: TypeError: Cannot read property 'implicitHeight' of null
2025-03-05T20:26:44.815 Debug: os: linux
2025-03-05T20:26:44.815 Debug: desktopAvailableWidth: 3286
2025-03-05T20:26:44.815 Debug: desktopAvailableHeight: 1080
2025-03-05T20:26:44.815 Debug: orientation: 2
2025-03-05T20:26:44.815 Debug: devicePixelRatio: 1
2025-03-05T20:26:44.815 Debug: pixelDensity: 3.9751024219785513
2025-03-05T20:26:44.815 Warning: qrc:/QML/MainView.qml:931: TypeError: Cannot read property 'qQStyleTheme' of null
2025-03-05T20:26:44.815 Debug: Creating component: qrc:/QML/RadioView.qml
2025-03-05T20:26:44.869 Debug: Creating component: qrc:/QML/MotView.qml
2025-03-05T20:26:44.921 Warning: QQmlComponent: Created graphical object was not placed in the graphics scene.
2025-03-05T20:26:44.924 Warning: qrc:/QML/MainView.qml:674: TypeError: Cannot read property 'device' of null
```

Any advice?

[image]

andimik, 2025-03-05:
I found another way.

```
git diff
diff --git a/src/welle-gui/QML/GeneralView.qml b/src/welle-gui/QML/GeneralView.qml
index 033f3d42..8ec2f398 100644
--- a/src/welle-gui/QML/GeneralView.qml
+++ b/src/welle-gui/QML/GeneralView.qml
@@ -26,6 +26,7 @@
 import QtQuick
 import QtQuick.Layouts
 import QtCore
+import Qt.labs.settings 1.1
 
 import "components"
 
diff --git a/src/welle-gui/QML/MainView.qml b/src/welle-gui/QML/MainView.qml
index 1ca61550..657960ab 100644
--- a/src/welle-gui/QML/MainView.qml
+++ b/src/welle-gui/QML/MainView.qml
@@ -30,6 +30,7 @@ import QtQuick.Controls.Universal
 import QtQuick.Window
 import Qt5Compat.GraphicalEffects
 import QtCore
+import Qt.labs.settings 1.1
 
 import "texts"
 import "settingpages"
diff --git a/src/welle-gui/QML/settingpages/ChannelSettings.qml b/src/welle-gui/QML/settingpages/ChannelSettings.qml
index 89b9aec6..b5b03702 100644
--- a/src/welle-gui/QML/settingpages/ChannelSettings.qml
+++ b/src/welle-gui/QML/settingpages/ChannelSettings.qml
@@ -27,6 +27,7 @@ import QtQuick
 import QtQuick.Controls
 import QtQuick.Layouts
 import QtCore
+import Qt.labs.settings 1.1
 
 import "../texts"
 import "../components"
diff --git a/src/welle-gui/QML/settingpages/ExpertSettings.qml b/src/welle-gui/QML/settingpages/ExpertSettings.qml
index ffe7122e..edc5da91 100644
--- a/src/welle-gui/QML/settingpages/ExpertSettings.qml
+++ b/src/welle-gui/QML/settingpages/ExpertSettings.qml
@@ -27,6 +27,7 @@ import QtQuick
 import QtQuick.Controls
 import QtQuick.Layouts
 import QtCore
+import Qt.labs.settings 1.1
 
 import "../texts"
 import "../components"
diff --git a/src/welle-gui/radio_controller.cpp b/src/welle-gui/radio_controller.cpp
index 9520e94d..f8031989 100644
--- a/src/welle-gui/radio_controller.cpp
+++ b/src/welle-gui/radio_controller.cpp
@@ -891,7 +891,7 @@ void CRadioController::displayDateTime(const dab_date_time_t& dateTime)
 
     // int OffsetFromUtc = dateTime.hourOffset * 3600 +
     //                     dateTime.minuteOffset * 60;
-    currentDateTime.setTimeZone(QTimeZone::UTC);
+    currentDateTime.setTimeZone(QTimeZone::utc());
 
     emit dateTimeChanged(currentDateTime);
 }
```

but I needed to install 

```
sudo apt-get install qml6-module-qt-labs-settings 
```

It will start the app without commenting out the `Settings`, but I still get seg fault ...

AlbrechtL, 2025-03-07:
Please try compiling against Qt 6.9.

andimik, 2025-03-08:
Then you should remove all `sudo apt-get install` commands from the Readme and mention the minimum Qt version which is required.

I am not lucky that latest Ubuntu LTS is not supported out of the box.

closed 2025-03-08 by andimik

AlbrechtL, 2025-03-08:
To update the readme.md is in my todo list. Before I will release the version 2.7 I will update it. Sorry for the inconveniences!

> I am not lucky that latest Ubuntu LTS is not supported out of the box.

You are talking about the packages that are coming from the standard Ubuntu repository. Here is the thing, the included software is usually not up to date. From a developer perspective, that's bad because you have to wait years to use new library features when relying on only the repo packages. I personally use Ubuntu LTS for development, but with the newest Qt libraries that can be downloaded directly via https://www.qt.io/download-qt-installer-oss. Especially talking about Qt there is a lot of background development supporting Wayland, high DPI displays, Windows 11, arm64 support, etc. Because my personal time that I can spend to welle.io is really limited, I decided to support only the latest Qt version for development.

andimik, 2025-03-08:
Ok, understood. Thanks for the effort and for the explanation.