AlbrechtL/welle.io #850: Flatpack under Linux misses librtlsdr.so.0

andimik, 2025-03-12:
```
flatpak run io.welle.welle-gui 
welle-io: error while loading shared libraries: librtlsdr.so.0: cannot open shared object file: No such file or directory
```

The problem appeared today.

Older flatpack versions will work.

```
flatpak list 
Name                             Anwendungskennung                           Version                       Zweig             Ursprung                Installation
welle.io                         io.welle.welle-gui                          2.7-unstable-fc9c89f0         master            welle-gui-origin        user
welle.io                         io.welle.welle_io                           2.7-unstable-c99d2184         master            welle_io-origin         user
Mesa                             org.freedesktop.Platform.GL.default         24.3.4                        24.08             flathub                 user
Mesa (Extra)                     org.freedesktop.Platform.GL.default         24.3.4                        24.08extra        flathub                 user
openh264                         org.freedesktop.Platform.openh264           2.4.1                         2.4.1             flathub                 user
Mint-Y-Aqua Gtk Theme            org.gtk.Gtk3theme.Mint-Y-Aqua                                             3.22              flathub                 user
KDE Application Platform         org.kde.Platform                                                          6.8               flathub                 user
```

AlbrechtL, 2025-03-13:
Oh boy! My working build flatpak was working. Thanks for reporting!

AlbrechtL, 2025-03-13:
Fixed in https://github.com/AlbrechtL/welle.io/commit/7e7b1b68cadfbcbc8ad338c1c2cade46de0a6851.

closed 2025-03-13 by AlbrechtL