AlbrechtL/welle.io #854: PPP-RTK-AdV - GPS RTK Data - Support dumping packet-data for service

flohoff, 2025-04-07:
Hi,

In 01/2025 the German Cadastre Offices (Arbeitsgemeinschaft der Vermessungsverwaltungen der Länder der Bundesrepublik Deutschland) officially started broadcasting GPS RTK (Realtime Kinematic) informations in the "DR Deutschland" Ensemble.

It is visible as 

`  [0xe0d210bc] PPP-RTK-AdV       [component 0 ASCTy: DAB ] [subch 32 bitrate:8 at SAd:468]`

See https://gepos.sapos.de/wp-content/uploads/2025/02/202502_Produktinformation_PPP_RTK_v1_3.pdf

> DAB+-Kanal 5C (1. Bundesmux),
> Subkanal 32 Bezeichnung "PPP-RTK-AdV",
> Frequenz 178,352 MHz

I know that packet/data handling is currently missing in welle.io. I saw the "pseudocode" for DabData in the msc-handler.

I stripped down welle.io so i could build a welle-rtk binary without any Audio dependencies/decoders. I am at the point that i am trying to inject a DabHandler into the msc-handler stuff.

Just the question whether there is a branch somewhere containing some code, or someone else is already working on packet data or even the PPP RTK stuff. Probably just as a sync code for others.

Flo

AlbrechtL, 2025-04-07:
@flohoff Great initiative!

You can check https://github.com/JvanKatwijk/qt-dab (actually, welle.io is a fork of qt-dab). There is some non-audio code included, but I don't think for  PPP RTK.

If you have working code, feel free to open a PR.

flohoff, 2025-04-07:
I had a look at qt-dab and there is adv code - but it seems only some demux code based on the stream type. I cant see any output code to anywhere. And in the end this needs to be fed to RTKLIB - and it needs more handling decoding.

mpbraendli, 2025-04-08:
Hi and thanks for opening the discussion here after our exchange on the osm forum.

I don't know what the best interface towards RTKLIB is, so please make a proposal.

I'm ok with having a dedicated "welle-extract-PPP-RTK-AdV" binary with reduced build dependencies and a single well-defined purpose, and I'd love to have a HTTP endpoint in welle-cli for data subchannels. But we can do that later.

I haven't checked recently, but I don't think qt-dab handles "Transparent Data Channel in a packet mode service component without data groups" (TS 101 759 Clause 4.1.1)

flohoff, 2025-04-08:
I am at the point that i see the packets on the Subchannel as they come in with and without FEC.

Thats a 24 byte byte packet with address set to 0x3fe which is described in 5.3.5.2 - Address shall be 1022 aka 0x3fe. 

```
PCount: 94
	Length: 0
	Counter: 7
	Address: 03fe (FEC)
	Data:  1f fe 54 91 ed 9c d9 ad 12 5c 9c 87 b4 e7 c6 a6   ..T......\......
	Data:  b7 1f ff e5 51 4d cf a2                           ....QM..        
```

On the very same subchannel i do see other packets aswell - thats the ones without FEC. Currently i distinguish them by their address - and for the non FEC case i check the CRC. I havent seen a "better" way ...

```
PCount: 86
	CRC: 1203 (Valid)
	Data:  00 02 13 36 cc c9 4e 49 ef 59 72 fb 9f 60 7f 88   ...6..NI.Yr..`..
	Data:  74 fa ae 9f 2b 2f 12 03                           t...+/..        
```

Next step would be to reassemble the non-FEC packets, and try Reed Solomon on the FEC ones. In 5.3.5.3 the considerations says that a receiver may use the non-FEC packets and only switch to FECs ones in case of receiption issues.
I am unshure whether thats possible without application knowledge of the packet stream though.

I am working here.

https://github.com/flohoff/welle.io/tree/rtk

Some stuff is nice to have - as removing DabAudio from the msc-handler 5d60830cadb03b859c6be547189c86a76bde6b66 which makes it possible to build a binary without DabAudio. 

10eb5ddd6a68d0d47f19dcddf5730a3611fec5c0 is a nice to have for the de-interleaver to use only one constant and modulo and not "& 17" and in other places < = 15  - makes it hard to read.

And in the end i'd like to the deinterleaver, deconvolve and dedisperse be more capsulated aways so we can reuse the stuff. Its basically the same for ALL streams not just audio.
 
Flo