AlbrechtL/welle.io #855: QT6.8 dependency / QTimeZone::UTC

flohoff, 2025-04-07:
Hi,
while trying to get some of the hacks into shape i found that in welle-gui/radio_controller.cpp you are using QTimeZone::UTC which is QT6 6.8. 

I am on Debian/Bookworm which only has 6.4 so this blows up.

This was the quick hack - I am unshure whether this is performance critical or mem leaky.

https://github.com/flohoff/welle.io/commit/abecc80faadc916bce807cac846566b24440d32e

Flo

andimik, 2025-04-07:
https://github.com/AlbrechtL/welle.io/issues/845

I also have faced the same problem. Ubuntu 24.04 is not supported out of the box.

AlbrechtL, 2025-04-08:
Thanks for reporting.

Indeed, Qt 6.4 is not supported. Great that you got it working with such a small patch. Since time is going on, this issues will disappear in the future.

closed 2025-04-08 by AlbrechtL