AlbrechtL/welle.io #861: Streaming a local IQ file or a distant IQ file

f4iop, 2025-06-17:
Hello Albrecht, hello all,

i am teaching DAB+ since 15 years in France (INA Institut National de l'Audiovisuel) and i am actually playing with hundreds hours (more than 150 TB) of very large IQ files recorded in mobile over France. We do have global IQ files (55 Ms/s 50 MHz.BW) to get a big part of the VHF band, and have the IQ extract file of each DAB+ channel from 5B to 11D range or other ranges depending the geography.

Then we have IQ files sampled @ 2.048 Msamples/s, 2.048 MHz Bandwidth, 16 bits little indian for each channel. These files are readable by AbracaDABra software, but i would dream Wello.io to perform such file reading also because we can add some log output by recompiling welle.io for our special expectations.

The target is to check the SFN sync of transmitters by an offline line processing, in areas where these transmitters are expected to be synced.

If you think about implementing the IQ file reading, this can be thought as internet file reading (via an url) also. We know that such a stream would require a high download bandwidth (130 Mbits/s for 16 bits IQ, or half for 8 bits IQ) but these values begin to be acceptable.

The 5 bytes header known to be found on the rtl_tcp protocol could be perhaps maintained.

Kind regards,

Gilles 
F4IOP

f4iop, 2025-06-17:
Well, hum...

i tried to read the following IQ file, 16 bits, little indian, 15 minutes 

http://gilles.misslin.com/trial_6B.raw (7.7G)

=> Welle.io : time to time, less 1% of the time, Welle.io is able ti decode something
=> abracaDABra : is able to decode it perfectly with 0 errors all over its play

Strange...

Screen copy with Welle.io
[image]

f4iop, 2025-06-17:
Screen copies of abracaDABra :
[image]

[image]

andimik, 2025-06-17:
I'm glad to teach a teacher ;-)

Your file is in the wrong format, but this never minds, welle.io will play it as well. You need to select **s16be** instead of **s16le**.

Then you get all the TII codes (after enabling it in the options of course)

```
2025-06-17T22:44:04.804 Debug: TII comb  5  pattern  22  delay  -1 =  -0.146484  km  with error  205
2025-06-17T22:44:05.740 Debug: TII comb  5  pattern  22  delay  106 =  15.5273  km  with error  153
2025-06-17T22:44:06.683 Debug: TII comb  5  pattern  31  delay  106 =  15.5273  km  with error  166
2025-06-17T22:44:06.690 Debug: TII comb  5  pattern  28  delay  212 =  31.0547  km  with error  197
2025-06-17T22:44:06.695 Debug: TII comb  5  pattern  22  delay  106 =  15.5273  km  with error  188
2025-06-17T22:44:06.700 Debug: TII comb  5  pattern  34  delay  212 =  31.0547  km  with error  174
2025-06-17T22:44:06.703 Debug: TII comb  5  pattern  12  delay  212 =  31.0547  km  with error  182
2025-06-17T22:44:07.651 Debug: TII comb  5  pattern  22  delay  212 =  31.0547  km  with error  166
2025-06-17T22:44:08.600 Debug: TII comb  5  pattern  22  delay  106 =  15.5273  km  with error  175
2025-06-17T22:44:09.547 Debug: TII comb  5  pattern  22  delay  106 =  15.5273  km  with error  190
2025-06-17T22:44:10.485 Debug: TII comb  5  pattern  22  delay  106 =  15.5273  km  with error  167
2025-06-17T22:44:11.434 Debug: TII comb  5  pattern  22  delay  383 =  56.1035  km  with error  155
2025-06-17T22:44:11.437 Debug: TII comb  5  pattern  31  delay  106 =  15.5273  km  with error  163
2025-06-17T22:44:11.442 Debug: TII comb  5  pattern  28  delay  212 =  31.0547  km  with error  189
2025-06-17T22:44:11.446 Debug: TII comb  5  pattern  34  delay  212 =  31.0547  km  with error  176
2025-06-17T22:44:11.449 Debug: TII comb  5  pattern  12  delay  212 =  31.0547  km  with error  189
2025-06-17T22:44:12.477 Debug: TII comb  5  pattern  22  delay  -1 =  -0.146484  km  with error  159
2025-06-17T22:44:13.437 Debug: TII comb  5  pattern  22  delay  212 =  31.0547  km  with error  176
2025-06-17T22:44:14.382 Debug: TII comb  5  pattern  22  delay  -1 =  -0.146484  km  with error  170
2025-06-17T22:44:15.330 Debug: TII comb  5  pattern  22  delay  212 =  31.0547  km  with error  175
```

And the constellation diagram looks normal.

[image]

As far as I know Jan (the developer of AbracaDABra) included a check for 16be and 16le.

Of course, if your file would have an xml header (meaning you have created it with AbracaDABra, Qirx or Qt-DAB), then you could see it there.

f4iop, 2025-06-17:
Thank you so much Andimik,

well, i am not an expert about iq coding. I have began this project without any iq file handling and thanks to people like you, i get a better understanding :)

Kind regards !

andimik, 2025-06-17:
Well, I have technically no idea what is the difference (and why on earth there is no standardization), but I just played around with the options ...

f4iop, 2025-06-17:
"Then you get all some TII codes (after enabling it in the options of course). But they seem not to be complete in your region, compared to the screenshot of AbracaDABra."

=> i do not understand what do you mean by "not to be complete in your region" 

AbracaDABbra decodes perfectly the TII we can expect at the location where we record the IQ file. 
But the TII decoder of WELLE.IO (also noticed with a USB dongle receiver) show big incoherences.

f4iop, 2025-06-22:
Hello Albrecht and everybody,
I managed to find the issue with the filename extensions and make welle.io work in CLI mode with a file 
Here are appropriate examples that work fine !
test_6B.s16le.iq
test_6B.s16be.iq

closed 2025-06-22 by f4iop