AlbrechtL/welle.io #870: Audio decoding crashes the application

KMadura, 2025-07-19:
Almost everything works (scanning included) until I select a radio station to play, then the application crashes with following error in the console:

```
AACDecoder: using decoder 'FAAD2'
terminate called after throwing an instance of 'std::runtime_error'
  what():  AACDecoderFAAD2: error while NeAACDecSetConfiguration
```

OS: Debian Unstable (Trixie 13)
welle.io version in repository: 2.4+ds-4
libfaad2 version in repository: 2.11.2-1

I have tried following changes:
- Cloned welle.io repo, built welle.io, application has issues with SDR detection and can't connect to my Airspy - this problem doesn't occur on version installed from debian repository
- Cloned libfaad2 from https://github.com/knik0/faad2 , built it, welle.io returns the same error
- Any other changes to libfaad2 files (adding .bak to file name, removing .bak, running ldconfig all the time)

Is there anything else I can do?

AlbrechtL, 2025-07-24:
Thanks for the report.

To know what's going wrong, a deeper investigation is necessary. But before we dig into it, it would be worth to try the latest welle.io 2.7 version. You can install it via flatpak https://flathub.org/apps/io.welle.welle-gui or compile yourself (the airspy issue you discovered needs to be solved then).

AlbrechtL, 2026-10-11:
Thanks for the investigation. Feel free to raise a pull request.

AlbrechtL, 2026-10-13:
Please test it again in the latest next builds.