AlbrechtL/welle.io #873: Flatpak: Missing SoapySDR Modules

nemhods, 2025-08-17:
Hello all,

I was trying to get a HackRF One to work with the welle.io Flatpak, but it states that no valid device was found. I investigated the flatpak environment, and it seems like SoapySDR is installed, but none of the modules for hardware support.

Host system (Arch) after installing the Arch package `soapyhackrf`:
```
> SoapySDRUtil --info                                                                                                                                           
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.1-ARCH
API Version: v0.8.0
ABI Version: v0.8
Install root: /usr
Search path:  /usr/lib/SoapySDR/modules0.8
Search path:  /usr/local/lib/SoapySDR/modules0.8               (missing)
Module found: /usr/lib/SoapySDR/modules0.8/libHackRFSupport.so (0.3.4)
Available factories... hackrf
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]
```

Inside the flatpak:
```
> flatpak run  --command=sh   io.welle.welle-gui                                                                                                             
[📦 io.welle.welle-gui ~]$ SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.1-g1cf5a539
API Version: v0.8.0
ABI Version: v0.8
Install root: /app
Search path:  /app/lib/SoapySDR/modules0.8 (missing)
No modules found!
Available factories... No factories found!
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]
```

I'm not too familiar with flatpak best practices, but it seems like if we wanted to support multiple devices via Soapy, we would need to supply all Plugins in the flatpak, potentially increasing the file size and the number of dependencies.

I tried injecting `libHackRFSupport.so` into the flatpak via the `filesystem` switch. That almost worked, but it then complained about missing libhackrf libraries. I could chase this down and inject all libraries until it works, but this doesn't feel all too practical.

Is it just a matter of "not supported via flatpak"? Or am I doing something wrong?

AlbrechtL, 2025-12-28:
Fixed in https://github.com/flathub/io.welle.welle-gui/pull/4. Thanks to @AsciiWolf!

Copied `io.welle.welle-gui.yml` also in this repo see https://github.com/AlbrechtL/welle.io/commit/9e2792d2d8182619aa4823ae6ee7c80e3d6a6533.

closed 2025-12-28 by AlbrechtL

AsciiWolf, 2025-12-28:
I can test it with my HackRF One when time permits. :-)

We could probably also add USRP (UHD) and PlutoSDR support since both are very popular (although USRP is very expensive and not intended for home users) and should work out-of-box inside Flatpak on modern Linux systems. But I am not able to test these since I only own RTL-SDR, SDRplay, HackRF and Airspy SDRs.

AsciiWolf, 2025-12-29:
Sadly does not run:

```
$ flatpak run io.welle.welle-gui//master
welle-io: error while loading shared libraries: librtlsdr.so.0: cannot open shared object file: No such file or directory
```

AsciiWolf, 2025-12-29:
On the other hand, my Fedora system also has the rtl-sdr libraries in `/usr/lib64` and welle.io (the non-Flatpak version) works fine there.